open-ephys/bonsai-onix1 #19: Add configuration dialog for ONI context

glopesdev, 2023-08-29:
First iteration can adapt functionalities of the existing ONI Context GUI of the original package, accessing liboni directly.

glopesdev, 2024-07-02:
@jonnew I am assuming we are moving this to the 0.2.0 milestone.

jonnew, 2024-07-03:
Yes, I think thats reasonable.

jonnew, 2024-08-13:
During a meeting, we started discussing what this would actually look like and realized there are a lot of details to consider

- Should this configuration dialog actually access hardware?
- Should it reflect the state that _would be_ applied during the configuration chain? Should it apply the configuration chain? This is important because be default, e.g., the headstage ports are powered off. Does it need to turn them on to figure out the true device table or just tell the user what will be applied in the configuration chain if the devices are actually present when the play button is pressed.
- etc.

We are moving this to 0.3.0 for this reason.

aacuevas, 2025-05-19:
I believe this is one of these things that fall into the bonsai as a user frontend vs bonsai as a programming language, and the lib allowing access to things that *might* be useful for the latter.

However, it is true that we are not focusing too much, if at all, on that area. So I agree that this should be closed until there is a real need for a feature like this.

bparks13, 2025-05-19:
Yeah, this idea first came about by trying to emulate what was in `Bonsai.ONIX`, but we have since changed the infrastructure and the user experience so I don't think it's worth keeping. We can close this issue, and if we end up finding a new use case for it in the future we can create a new issue that is more relevant

glopesdev, 2025-05-19:
@jonnew the original issue seems to be talking only about a configuration dialog mimicking the old ONIX device table GUI. While I think this might still be potentially useful, even if only for low-level troubleshooting, I agree it is not vital.

However, I do feel the discussion which followed from this issue is still very relevant and I am not sure it is captured anywhere else? Is there an issue specifically about capturing and logging the device table as metadata?

If not, I think the discussion probably ought to be summarised and transferred to a new issue, or moved into the issue about generalised logging, as I really feel this problem is vital for data analysis.

I add a bit of motivation below, feel free to transfer it to a separate issue as appropriate.

Basically, as ONI adoption increases we will want to move away from hand-tailoring data access APIs and data writers for every particular combination of devices. ONI as a standard is fundamentally about providing the freedom to combine bio physiology acquisition systems. 

The natural expectation from such freedom is that it continues all the way to data logging and data analysis, and for this we will need a standard schema for data records of some sort. The device table is already a strong candidate to be the "standard" for this, since it is part of the ONI specification and therefore complete as far as describing the connected hardware.

The only thing missing is a way to map this hardware description language to whichever unique data storage "identifiers" are used for logging.

To achieve this, we can either create a new standard with guarantees of uniqueness, correctness, error handling, etc, or simply use the standard we already have (ONI) and simply dump the set of all raw data (frames) and corresponding metadata (device table).

The latter approach would enable building a full stack of Python/MATLAB APIs directly on the ONI standard to decode any combination of device data ever logged with any ONI system.

The user would not actually need to see any of the addresses and payload structures, but the logging and analysis system working together would keep track of this provenance and expose a standard view of what happened for analysis.

That power would be unrivalled when matched against anything you might hand craft with examples, since the latter will never cover all combinations of what people might want to use.

We did something similar for Harp and from what I have seen of ONI, it is already within reach, which is very exciting.

jonnew, 2025-05-23:
We can certainly leave this open, but I'm going to remove the milestone tag. In response to your message

> ONI as a standard is fundamentally about providing the freedom to combine bio physiology acquisition systems.

Yes, but, at least in my mind, really at the level of hardware interfacing. By way of comparison: we want a open hardware interface standard, API, agnostic driver interface that is better that DAQ-MX and its APIs .

> The natural expectation from such freedom is that it continues all the way to data logging and data analysis, and for this we will need a standard schema for data records of some sort.

ONI should be developed with this in mind, but really, its tailored for hardware. A good example of this is the datasheets. These could be turned into some formal schema describing device IO, but I feel like this would be a hinderance. The level of formality they currently have is what is generally used to describe digital hardware everywhere else and for good reason. The datasheets provide a complete, fairly formal description of hardware elements but are meant to be read and used by a human.

> The only thing missing is a way to map this hardware description language to whichever unique data storage "identifiers" are used for logging.

I wish that was the case, but I will use the current Neuropixels 1.0e implementation as a counter example. If you were to log raw frames, you would be logging data that is sow raw (pre calibration file application and ADC non-monotonic offset correction) that it would be de-facto wrong. What's worse is that since the _required_ calibration data is completely detached from the hardware producing the data, it would somehow need to be packed into any data file produced by this device. In this case, the data in ONI frames and described of the device by the device table is just not enough to accurately log the frame produced by the hardware. In this library, the output of the `NeuropixelsV1eData` operator is. 

I guess this does not describe the case as maybe we want it though. And perhaps part of the ONI standard could be that devices need to be self contained. AKA, unlike neuropixels, they cant rely on some magic external data. If that's true, it needs to somehow be injected into the hardware so that the device map and data frames are not incomplete when received by the host API. Or the device table could be made more extensible to allow the addition of data required to accurately extract data.

> or simply use the standard we already have (ONI) and simply dump the set of all raw data (frames) and corresponding metadata (device table).

Taking into account above comment, I think this is useful for sure and would result in the most compact representation of data produced by the system.  

Overall, of course I agree with the overarching idea you are presenting but I think there are major hurdles for implementation such that  is transparent to the end user.