open-ephys/bonsai-onix1 #35: Should an exception be thrown if StartAcqusition is not present

jonnew, 2024-01-16:
I was testing some things with the library, and I built this workflow. It ran without error. However, because its missing a **StartAcquisition** node, it didn't do anything. Is this a desirable behavior or should this workflow throw an exception indicating that Acquisition is not started?

[image]

glopesdev, 2024-05-28:
@jonnew We could throw on subscription if any nodes waiting on data detect that the context is not running. However, this would prevent more advanced scenarios where the start of acquisition is deferred until a specific moment.

My feeling is that the context not starting is a salient enough effect that it will be easily detected, since no data stream will be streaming, and we should not throw in this case.

jonnew, 2024-05-28:
Fair enough. I'm fine leaving this alone

closed 2024-05-28 by jonnew