open-ephys/bonsai-onix1 #53: Multiple devices of the same type on a hub create a naming conflict

bparks13, 2024-02-20:
If more than one device of the same type are in a hub, the automatic naming convention assigns the same device name to all devices of the same type.

Specifically, I am creating a hub that houses four **Test0** devices. Inside **HubDeviceFactory**, `UpdateDeviceNames()` only uses the *device.DeviceType.Name* when creating the device name, meaning that all four devices have the same name (e.g., `HubA.Test0`). 

Expected behavior would be unique device names (i.e., `HubA.Test0A`, `HubA.Test0B`, etc.).

This naming conflict shows up when I try to stream data from the `Test0Data` node, since I need to choose a specific device (e.g., `HubA.TestC`) from the pull-down menu, but only one option is available and throws an exception at runtime.

glopesdev, 2024-02-22:
@bparks13 The `UpdateDeviceNames` method is marked as `virtual` so specific hubs can override its behavior. The default behavior was meant to provide a convenient default, but in the meantime for specific hubs it can be changed to make everything explicit, for example:

```c#
class TestHub : HubDeviceFactory
{
    public ConfigureTest0 Test0 { get; set; } = new();
    public ConfigureTest0 Test1 { get; set; } = new();
    public ConfigureTest0 Test2 { get; set; } = new();
    public ConfigureTest0 Test3 { get; set; } = new();

    internal override void UpdateDeviceNames(string hubName)
    {
        Test0.DeviceName = $"{hubName}/{nameof(Test0)}";
        Test1.DeviceName = $"{hubName}/{nameof(Test1)}";
        Test2.DeviceName = $"{hubName}/{nameof(Test2)}";
        Test3.DeviceName = $"{hubName}/{nameof(Test3)}";
    }

    internal override IEnumerable<IDeviceConfiguration> GetDevices()
    {
        yield return Test0;
        yield return Test1;
        yield return Test2;
        yield return Test3;
    }
}
```

That said, I do agree that we can perhaps think of improving the default naming strategy to allow for both consistency and flexibility in device naming. The easiest choice would be to use the name of the device properties themselves to uniquely identify each device in the hub, which would ensure uniqueness, flexibility and consistency between property grid and device manager names.

Unfortunately you can't directly access the property names dynamically in code other than using reflection, but this is still definitely an option that we can consider.

Hopefully the above workaround will suffice to unblock development for now, but let's maybe discuss this next week and I will try to come up with a few implementation alternatives.

glopesdev, 2024-07-02:
@jonnew I think we discussed previously that we were happy with the current approach. We can definitely improve the API for future versions if this becomes a more common pattern, but for now I think this is not really a blocker so will close the issue.

closed 2024-07-02 by glopesdev

jonnew, 2024-07-03:
Yes very happy with it.