open-ephys/bonsai-onix1 #58: Issue with setting hs-64 electrical stimulator enable register

jonnew, 2024-03-05:
In [3efaf1b](https://github.com/neurogears/onix-refactor/pull/46/commits/3efaf1be66347fb94c7edc08d1a5ccd4af03111a) I added basic functionality for the electrical stimulation sub-circuit within headstage-64. However,  the trigger node is non-functional because the ENABLE register state is not maintained at 1 after its configured. Namely,

[image]

This is a Configuration property that is set a single time when creating the observable. I have a strong feeling that this is due to ENABLE's reset behavior, which is a bit different than other registers, but is valid under the spec:

[image]

(Link to complete register table: https://open-ephys.github.io/onix-docs/Hardware%20Guide/Datasheets/estim-hs64.html#register-programming)

The highlighed value is the value that the ENABLE is set to on a reset. The POWERON  register is shown for comparson, where None indicates that the reset has no action on the its value. Most registers behave like POWERON. But for something like enabling the electrical stimulator, its important that the register always start fresh at "off". Maybe this just indicates that ENABLE really needs to be an Acquisition-level, dynamic property that is set to true as part of the workflow's logic at runtime.

glopesdev, 2024-03-31:
> Maybe this just indicates that ENABLE really needs to be an Acquisition-level, dynamic property that is set to true as part of the workflow's logic at runtime.

@jonnew What we have been doing now in similar cases is to have `Configure` operators which can be used to set both initialization time and runtime configuration parameters (see [modify-pulse-pal-configuration](https://bonsai-rx.org/pulsepal/index.html#modify-pulse-pal-configuration)), and also to allow dynamic setting of individual properties (instead of having everything grouped inside a single configuration or runtime node.

What I like about this is that often for dynamic stimulation scenarios you want to start with a stable known config, and then change it as part of the experimental logic, *but without overwriting your original settings*. By having everything going through only one node with externalizable properties, there is a significant risk of someone saving the workflow after running the protocol and losing important settings.

The other nice thing about this approach is that it is possible to design the architecture to make the code completely reusable, so you wouldn't have to rewrite the logic to allow for these different options.

I can show you an example of this next time we meet.

glopesdev, 2024-04-23:
Fixed via #66

closed 2024-04-23 by glopesdev