open-ephys/bonsai-onix1 #60: Namespaces, packages, Nuget naming

jonnew, 2024-03-12:
We need a coherent Namespace/package categorization and naming convention. We want different hardware to be split up into different packages. However, there is a lot of code reuse between hardware. Further, we want to follow existing package and namespace naming conventions. MS [recommends](https://learn.microsoft.com/en-us/dotnet/standard/design-guidelines/names-of-namespaces?redirectedfrom=MSDN) the following:

`<Company>.(<Product>|<Technology>)[.<Feature>][.<Subnamespace>]`

What is Product/Technology here?

1. The targeted technology (Bonsai)?
2. The spec (ONI)?
3. The specific hardware (Headstage64)?
4. Some combination of them?

What do we do with reused, highly-specific code? e.g. `ConfigureBNO055`?

Please put your proposals below.

jonnew, 2024-03-12:
Goncalo suggests the following:

OpenEphys.Oni (clroni)
OpenEphys.Oni.Bonsai (core bonsai lib)
OpenEphys.Onix (reusable ONIX components)
OpenEphys.Onix.Headstage64 (example ONIX product)

Jon pretty much agrees with it.

bparks13, 2024-04-23:
We have created a workflow, utilizing Github Actions, to automatically build/test packages before uploading to NuGet. This is shown in [OpenEphys.Sockets.Bonsai](https://github.com/open-ephys/OpenEphys.Sockets.Bonsai). 

The general flow is to create a branch or a fork of the repository, make changes, then open a PR to the main branch. After opening the PR, actions are automatically run that verify a version number is increased, and the package is built correctly (i.e., without errors). This is enforced by a number of rulesets that protect the main branch from direct commits, require the test/build to pass before merges can happen, and also enforce a linear commit history to make it easier to go back to a previous version if necessary.

Any updates or recommendations are welcome.

jonnew, 2024-04-23:
We also need to define conventions for naming nodes that produce and accept data

- Right now, most nodes that produce data have Data suffix, but its not consistent
- I prefer low level types to have a name like Bno055 -> Bno055Hardware and the data stream is named Bno055 without Data suffix. 

In any case, we need a consistent scheme.

glopesdev, 2024-06-18:
@jonnew @aacuevas To follow-up on this I think what we need to agree on now is a partition of which devices go into the common package, and which go into the headstage-specific packages.

Below is a working draft proposed breakdown:

### Legend
📦 (package)
🖲️ (headstage / hub)
⚙ (device)

### Proposed Breakdown
📦OpenEphys.Onix
 ┣ 🖲️BreakoutBoard
 ┣ ⚙FmcLinkController
 ┣ ⚙HarpSyncInput
 ┣ ⚙DS90UB9x
 ┣ ⚙AnalogIO
 ┣ ⚙Heartbeat
 ┣ ⚙LoadTester
 ┣ ⚙MemoryMonitor
 ┗ ⚙Test0
📦OpenEphys.Onix.Devices
 ┣ ⚙Rhd2164
 ┣ ⚙Bno055
 ┗ ⚙TS4231
📦OpenEphys.Onix.Headstage64
 ┣ 🖲️Headstage64
 ┗ ⚙Headstage64ElectricalStimulator
📦OpenEphys.Onix.NeuropixelsV2
 ┣ 🖲️NeuropixelsV2eHeadstage
 ┣ 🖲️NeuropixelsV2eBetaHeadstage
 ┣ ⚙NeuropixelsV2e
 ┣ ⚙NeuropixelsV2eBeta
 ┣ ⚙NeuropixelsV2eLinkController
 ┗ ⚙NeuropixelsV2eBno055

jonnew, 2024-06-18:
### OpenEphys.Onix.Devices
So the idea for `OpenEphys.Onix.Devices` is to place common stuff. In one sense I agree, in another sense I don't.

- BNO055: Basically agree
- RHD2164: gateware implementation is super old and limited. There are headstage-64's in the field that use it, but if we make another FPGA-based headstage with this chip, we will probably change some things that would result in a major revision. For gateware this means a different device entirely, that is aimed at the same physical chip. 
- Heartbeat: We plan on adding this to all FPGA based headstages, feels like it should go in this library and that `OpenEphys.Onix` should depend on it.

### OpenEphys.Onix 
- Breakoutboard is a beta test device indicating that this package should be called **OpenEphys.Onix.FMC**
- Loadtester, Test0, and Heartbeat are all generic and used in places other than base system

**The take home:** I see what you are doing, but to be honest, I think that common Devices should be in their own repo, which is never transformed to Nuget but instead, is brought into packages that need the devices as submodules. This will prevent the scattering of devices all across nuget packages and will provide some coherence in terms of repo locations.

glopesdev, 2024-06-18:
@jonnew This all sounds good, I have updated the above proposal to reflect your suggestions, and I have removed the `Devices` package. My only remaining question is around the `FmcLinkController` but we can discuss this in a bit and come back to update the breakdown.

glopesdev, 2024-06-18:
A discussion which came up is how to organize the upcoming NeuropixelsV2 which does not depend on the passthrough device in this package organization. One proposal is to make all the "e" devices be dependent on `FmcHost` (or another name we come up with for this host) and incorporate that into the package namespace, e.g. `OpenEphys.Onix.FmcHost.NeuropixelsV2` would be a package containing all passthrough NeuropixelsV2 devices.

aacuevas, 2024-06-18:
Regarding the last message, the rationale is this: Right now devices that follow the oni spec work perfectly across different hosts (e.g.: the bno055, which I've used on the usb acq. board). However, anything that is passthrough-based is tightly tied to the current host.
Our current plan is to deprecate those, enforcing every device to be a proper oni device (however the hardware wants to accomplish this, in onixv2 with the modules, for example), so those would be normal devices that can be in their normal namespaces. But current passthrough devices, for clarity, we think might be better if tightly tied to the host that they required.

glopesdev, 2024-06-18:
An alternative breakdown is to just keep the current mono-package and rename it to `FmcHost` to imply that all devices depend on the FMC architecture.

### Alternative Breakdown
📦OpenEphys.Onix
📦OpenEphys.Onix.FmcHost
 ┣ 🖲️BreakoutBoard
 ┣ ⚙FmcLinkController (?)
 ┣ ⚙HarpSyncInput
 ┣ ⚙DS90UB9x
 ┣ ⚙AnalogIO
 ┣ ⚙DigitalIO
 ┣ ⚙MemoryMonitor
 ┣ 🖲️Headstage64
 ┣ ⚙Rhd2164
 ┣ ⚙Bno055
 ┣ ⚙TS4231
 ┣ ⚙Headstage64ElectricalStimulator
 ┗ ⚙Headstage64OpticalStimulator
 ┣ 🖲️NeuropixelsV2eHeadstage
 ┣ 🖲️NeuropixelsV2eBetaHeadstage
 ┣ ⚙NeuropixelsV2e
 ┣ ⚙NeuropixelsV2eBeta
 ┣ ⚙NeuropixelsV2eLinkController
 ┗ ⚙NeuropixelsV2eBno055

jonnew, 2024-07-26:
I think I'm leaning towards:

**OpenEphys.Onix.PCIe**

Our next gen system is also PCIe, but it does this over an IPASS cable, so that one probably would be OpenEphys.Onix.IPASS or more locally devided into different packages as we have discussed above.

glopesdev, 2024-07-26:
@jonnew the spelling itself is fine, just as you say there might be other PCIe devices in the horizon. More importantly, because ONI is a protocol, it could give the wrong impression that this is all of ONI(x) over PCIe, which it is, but not in that general sense yet, so I wonder if this could bring other problems down the line if we reserve such a general namespace.

I am also thinking about namespace hierarchy. Down the line the modular architecture is still the goal so having `OpenEphys.Onix` as the core namespace and having sub-namespaces for different "families" makes sense. However, there is no modular `OpenEphys.Onix` yet, so there are advantages in considering the current system as a self-contained product which is modular within its own product line.

Because of the above I would actually not use `OpenEphys.Onix` at all yet and instead would form a compound word. My current preference would be something like:

- `OpenEphys.OnixBreakout`
- `OpenEphys.OnixV1`

jonnew, 2024-07-27:
After sleeping on it, i think the name should be:

`OpenEphys.Onix1`

Its the first in what i envision will be a collection of onix acquisition systems.

closed 2024-07-27 by jonnew

glopesdev, 2024-07-28:
@jonnew Sounds good, my only clarification question is to confirm the capitalisation.

My original suggestion was `OniX1` to emphasise that `Oni` is the standard and `X1` is the system model name. I like it better than leaving the `1` in isolation and also means people can refer to "the X1 system", "the X2 system" etc, but maybe that creates confusion with the phonetics of reading the name.

The best example of similar name with or without X and numbers:

- DirectX 12 (DX12)
- Direct3D 12 (D3D12)