open-ephys/bonsai-onix1 #88: NeuropixelsV2 channel configuration GUI

glopesdev, 2024-05-07:


jonnew, 2024-08-12:
I'm providing feedback on issue-88 @ 09a936b997 here:

- In the electrode selection GUI, i can zoom to a point where the probe is totally off screen. Is it possible to put some boundary that prevent the user from getting into a state where there is nothing by white on the screen?
- In the electrode selection GUI, I can zoom out to a point where the probe is just a tiny dot. Can we limit the zoom to the default zoom level in which the whole probe is visible
[image]
- The electrode selection does not respect Cancel button. If I select some new electrode configuration, then hit cancel, when I reopen the GUI the electrode selection that I attempted to cancel is still applied
- The bottom of the scrolling tool is behind the button frame of the GUI:
[image]
- I suggest the following reorganization for GUI elements:
[image]
  - The slider takes up the left  of the property pane to link it to the drawing. It should scale to take up the whole height of the drawing (unlike my red box).
  - The presets selection goes under the reference selection using the same layout
  - The select and deselect buttons go next to each other under that
- Please provide keyboard shortcuts for select and deselect. I suggest Enter and Delete. Make this information using a ToolTip when the user hovers over the select and deselect buttons

bparks13, 2024-08-12:
@jonnew Thanks for the feedback! I'll get to work on it now. I have a couple of clarifying questions in the meantime.

1. There is already (or there should be) limits on how far the user can zoom in; are you describing the edge case where zooming in repeatedly at this boundary can lead to a white screen? Or something else?
2. Yes, we can add a limit to the zoom out
3. When you say you "select" a new configuration, do you mean clicking and dragging to select a potential new set of channels, or is this coming from choosing a preset configuration from the drop-down? Either way this behavior shouldn't happen
4. Yes, this is a known issue due to resolution and scaling, and is already being addressed. Additionally, there is a button that cannot be seen on that screen at the bottom right that allows the user to reset the zoom level to the default, which will be moved up.
5. This is a reasonable reorganization. Related to 4, the previously hidden Reset Zoom button can be added to this grouping of controls so it is not an orphan at the bottom of the screen
6. Agreed, Enter is a good option for enabling. For deselect, either Delete or Backspace I think are appropriate, and maybe even both if there is no reason we might need these hot keys in the future

jonnew, 2024-08-12:
1. If you put your mouse way off to the side and then scroll, you can zoom to a point that no longer contains any electrodes
3. Either, from my recollection
5. Good -- i could not see the button on mine. Just move it close to the tohers
6. keep it simple, i think delete is probably enough.