open-ephys/bonsai-onix1 #103: Consistent naming convensions

jonnew, 2024-05-31:
I think its important to have consistent naming conventions for different types of nodes in the library. Currently `Configure<whatever>` nodes can go in the "configuration chain" at the top of the workflow. This seems to be pretty (if not completely) consistent. However, this is not true for data nodes. Most are named `<whatever>Data`, but there are breaks from this convention. For instance `AnalogInput ` and `AnalogOutput`.  TBH, I never really liked the `<whatever>Data` convention and perhaps the solution in this case is just to default to the device names for data producer/consumer nodes (e.g. `Bno055Data` -> `Bno055`). This of course will result in a conflict because of the static classes defining registers etc., but these are completely hidden from the user and therefore could stand to have less simple names. If we wen this way, nodes that don't map 1-1 onto devices (e.g. `AnalogInput` and `AnalogOutput`) can remain as is.

glopesdev, 2024-05-31:
@jonnew I don't like giving the data nodes the names of the devices, mostly because they are really not devices. If they were you would expect to be able to configure them, do both input/output, etc, similar to what happened in the old API.

Right now these nodes for the most part really represent the "data streams", and specifically mapping directly to FrameReceived streams.

However, we also already have cases where we have both "Data" and other nodes, like in the case of Rhs2116 where we have Rhs2116Data and Rhs2116Trigger which have totally different functions. I don't think any of these two nodes deserves to be named after Rhs2116 device itself since none of them will ever encapsulate its full functionality.

I do agree that "AnalogInput" and "AnalogOutput" are inconsistent. I can explain the logic behind the name: the actual device in this case is named "AnalogIO" which is abbreviation for "AnalogInputOutput". Each of these two nodes really encapsulates only one of the directions, but it seemed really awkward to call it "AnalogIOInputData" and "AnalogIOOutputData" so I opted to drop the "IO" from it.

I did drop the "Data" suffix at the time for reasons of aesthetic preference, but we could just call them "AnalogInputData" and "AnalogOutputData" and it might be mostly consistent?

glopesdev, 2024-07-02:
@jonnew I think we discussed this in a previous meeting and we were mostly fine with the current naming conventions for `AnalogIO` and the rest of the operators, especially given that this package will now be self-contained to the FMC host family of devices, so I am closing this issue.

We can revisit this in future packages, but mostly assuming we have good namespace scopes we should be able to find good tradeoffs for most names.

jonnew, 2024-07-03:
I think my opinion has not changed: things that accept or produce data should either be named as if they are hardware producing the device. So i like "AnalogInput" and I don't like "Bno055Data". However, given the huge effort this would entail when it comes to conflicting with static class names etc, I think a compromise is to actually double down on the current system in order to make it consistent. That means what few outliers that produce frames but don't have "Data" at the end of their name would have that added

AnalogInput -> AnalogInputData
AnalongOutput: No rename.

I would rather have "AnalogInputData" consistent with all the other nodes that produces frames compared to having it be symmetric with "AnalongOutput".

To address the RHS point: "Rhs2116Data and Rhs2116StimulusTrigger which have totally different functions"

They have totally different functions because one is a stream produced by an RHS2116 chip and the other is piece of HDL on the FPGA sitting next to it that governs synchronization between two RHS2116 chips. So perhaps the issue is that the trigger is not named correctly from the beginning (although it is highly specific to this particular function and has no chance of reuse elsewhere). 

Its interesting how this discussion about naming ends up reflecting the deepest levels of the hardware design. Easily following naming patterns seems to co-occur with well defined, non-muddled hardware designs.

glopesdev, 2024-07-03:
> Its interesting how this discussion about naming ends up reflecting the deepest levels of the hardware design. Easily following naming patterns seems to co-occur with well defined, non-muddled hardware designs.

100% agree with this, which is why naming is probably the hardest problem in all computer science, with cache invalidation a close second 😄 

> AnalogInput -> AnalogInputData

This still doesn't quite make it consistent since `AnalogInput` is not really a device so it can't have `AnalogInputData`. The device is called `AnalogIO` but calling this `AnalogIOData` also is very confusing IMO, which is why I was left scratching my head.

I agree this problem is not finished so let's leave the issue open. However, given the "beta" nature of the entire FMC hardware *and* software, I would maybe not even rename anything from what we have now and just leave it as an explicit example of what we don't like so we can look hard at it when we redesign the next device family.

aacuevas, 2024-07-04:
I'm thinking, should not we call this device BreakoutAnalog or similar? Since it is indeed an analog io that is tied to the current system. The new one might use a different device

closed 2024-08-13 by jonnew