open-ephys/bonsai-onix1 #106: Headstage Neuropixel 1.0e - GUI

bparks13, 2024-06-05:
Closely related to #100, and is focused on adding GUI elements on top of the work done in the issue-100 branch.

jonnew, 2024-06-12:
I cannot checkout and compile issue-106 branch because its missing the probe interface dependency. @glopesdev what do you do when you have some project dependency that is in active development and not yet ready for nuget but you dont want to put into the source of the repo that uses it?

bparks13, 2024-06-12:
The project dependency is present, can you try running the following code to pull the ProbeInterface to your local copy and see if that works?

```shell
git submodule init
git submodule update
```

This is one of the bigger problems with submodules, from what I know, is that you cannot automatically clone down the repo and also clone down the submodules at the same time. There are commands that can be used in Git Bash to pull the submodules in at the same time, but programs like VSCode do not do this automatically. If there is a way to set it so that the submodules are pulled when the rest of the repo is cloned down that would be great!

jonnew, 2024-06-13:
Shoot i didnt notice the submodule, ill try this today. Thanks @bparks13

PR #231 (2024-08-22): Add GUI for Neuropixels V1e
This PR aims to add a dialog GUI for the `NeuropixelsV1e` headstage. 

Controls are based on `NeuropixelsV2e` GUI, for more info on the controls see #214.

One major difference between V1 and V2 is that V1 has two separate calibration files that are required (Gain and ADC). The probe will not be visible unless both files exist and are selected.

Closes #106