open-ephys/bonsai-onix1 #116: Improve exception messages

jonnew, 2024-06-25:
This message is not helpful

https://github.com/neurogears/onix-refactor/blob/481a9150953d1893c916aebb1b0dda721362d527/OpenEphys.Onix/OpenEphys.Onix/ContextHelper.cs#L17

because it results in this inscrutable text presented to the end user:

"The selected device is not a 33 device." 

For example. What was "selected"? What is a 33 device?

aacuevas, 2024-06-25:
I am not sure if this should be a subset of this or its own issue, but a version of this exception appears when a np1.0e headstage is plugged and configured and no probe is found (which is common given the flex connector). That message at the very least should be more concise.

glopesdev, 2024-07-02:
@jonnew This is why I didn't get rid of the [`DeviceID`](https://github.com/neurogears/onix-refactor/commit/3befb412405240bffd0d7f5610f73a12f4b4e68c) enum. If we were to pass as argument here the enum value instead of the number, we could get it's name directly from there using [`Enum.GetName`](https://learn.microsoft.com/en-us/dotnet/api/system.enum.getname?view=net-8.0).

Unfortunately this also means you are then stuck with maintaining a global register of names to device IDs in the core library, which may or may not be what we want (for this release would be fine anyway since it will be a mono-library anyway).

Another alternative is to pass in the type and assume there is a standard constant `ID` always with the same name that is accessed dynamically to get the device ID.

glopesdev, 2024-07-02:
@jonnew @aacuevas I noticed these functions in `clroni`: https://github.com/open-ephys/liboni/blob/dababf06f10449e4de0ce2c67176081d4f24342b/api/clroni/clroni/onix.cs#L110

Would it make sense to expose these as static methods so that we ask the driver itself to do the string conversion? At least that way we wouldn't have to maintain our own copy of the mapping from ID to name.

glopesdev, 2024-07-02:
@jonnew @aacuevas In the end decided to refactor context access to use types in the PR above. We can discuss pros and cons at our next meeting but after trying a number of different approaches I think this is the most reasonable compromise right now.

PR #125 (2024-07-16): Read device ID from type metadata
The current implementation of `GetDeviceContext` and `GetPassthroughDeviceContext` explicitly receives the device ID as a numerical argument. While functionally sound, this approach makes it harder to generate informative error messages, since there is no way to map the ID back to a device name.

This PR refactors this approach to instead receive the expected device type. The numerical device ID is then extracted directly from the type metadata by searching for a literal field with the name `ID`. If no such field is found, or a device with an invalid ID is det [... cut]
closed 2024-07-16 by jonnew