open-ephys/bonsai-onix1 #117: Headstage64 voltage setup timeout

aacuevas, 2024-06-25:
When a hs64 is connected by alone, the voltage search works as expected. If the chain contains a ´ConfigureNeuropixels1.0eHeadstage´ followed by a ´ConfigureHeadstage64´, the np1.0e voltage will be set correctly, hs64 voltage search will start (noticeable thanks to the leds) but it will fail with the same message as if no headstage was plugged (cannot lock), which suggests some kind of timeout.

Setting a fixed voltage to the hs64 fixes the issue.

aacuevas, 2024-06-29:
Just to confirm: another workaround, on this particular configuration (an hs64 a np1.0e) seems to be having the hs64 before the np1.0e on the chain.
I am not sure what the dependency is here, but voltage probing for each headstage should be independent of their positioning or of other headstages in the configuration chain.

glopesdev, 2024-07-02:
@aacuevas There is no explicit timeout, but there are `Thread.Sleep` calls in the port scanning implementation for HS64 do deal with the port voltage settle behavior that were tweaked over time to be as minimal as possible without compromising the scan behavior:

https://github.com/neurogears/onix-refactor/blob/5903bfcc06960db73b8c1c55987de0fbcdc07598/OpenEphys.Onix/OpenEphys.Onix/ConfigureHeadstage64.cs#L88

In the case of a direct port voltage override the default `Thread.Sleep` parameter is 500 ms, which is higher than the lower 200 ms we used for incremental port scanning.

Is there any way the port scanning behavior or NP1.0e might destabilize the HS64 port voltage ever so slightly that the 200 ms is no longer enough?

What I would recommend to debug this would be to change the port scanning wait time to 500 ms instead of 200 ms and see if that changes anything. Alternatively it would also be useful to know whether this is failing during the wait period in port scanning (L89), or during the last `CheckLinkState` call after voltages have been set (L101).

jonnew, 2024-07-04:
Just to be clear on the issue, does the hs-64 voltage search go through the whole process (which takes like 30 seconds) or does it fail immediately.

FWIW, in the specific case of hs-64 with this hardware, i really think the auto voltage search should be disabled completely. Its too sketchy and too long.

jonnew, 2024-08-13:
In the end, I think this is a wontfix situation

- We added the ability to override the voltage discovery algorithm on #105
- As noted in the code comments, this is really a hardware issue that is especially prevalent on hs-64. e.g. neuropixels headstages don't have nearly as much trouble.

closed 2024-08-13 by jonnew