open-ephys/bonsai-onix1 #123: Temporal, out-of-flow context infrastructure

aacuevas, 2024-06-29:
We have talked about this possibility many times, but I don't believe we have it written.

It might be interesting to have the ability to temporarily create a context to interrogate the hardware outside of normal workflow. Example cases for this are:
- Device table interrogation (useful for debugging purposes among others)
- Hardware version retrieval
- Probe connection checks (e.g. #95 )

Some of them might require more configuration steps than others, so it is not trivial to create a general case interface.
The important part is that these mini-context should be:
- Temporal: They would exist to perform a specific data retrieval and will close afterwards
- Out of workflow: They should not require a working workflow
- Blocking: Either they are short-lived (e.g. just a probe checking or a version check) or are tied to modal windows. In any case, these mini-contexts should be isolated and exclusive, so no possibility of two context clashing exisits.

glopesdev, 2024-06-29:
@aacuevas this should already be possible since ContextTask is a class that can be called from any code.

Is this feature targeting users or developers? Is it for some GUI? It might help to write up some concrete use case to help flesh out the proposal.

aacuevas, 2024-06-29:
Mostly for users, there are some quality-of-life operations that could benefit from this.

While ContextClass can be called from any code, we need to think a bit about the use cases and the context of calling the ContextClass ;) For example, while getting hw and fw version requires a simple context, checking NP seating as suggested by #95 requires proper hs configuration. I would like cases like that becoming muddy.

In any case, this is definitely out of the scope of 0.1.0, I would say is a rather low-priority for now, just wanted to have a register of it.

glopesdev, 2024-06-29:
If the use case is for self-contained headstage configuration GUI, and assuming the GUI is launched from the configure headstage node, then a context could be created inside the GUI and passed through that configure node itself to simulate "configuration".

Otherwise, I think these would have to be manually coded.

jonnew, 2024-08-13:
Related to #19

aacuevas, 2025-05-19:
I agree that this idea seems antithetical to how this library works now.

If anything, some of the listed use cases could be incorporated into a different application for system debugging (something similar to oni-repl), but definitely should not be the case for this library, as it is now.