open-ephys/bonsai-onix1 #127: Confirm scale and gain factor compensation bit depths for Neuropixels headstage

glopesdev, 2024-07-04:
@jonnew I found the following notes on an old repo clone and just wanted to make sure we ended up doing this correctly. Feel free to close if these have all been addressed.

```c#
// we want to read the config file
// - S16 (signed) word truncation after applying scale / gain factors (should we just truncate?)
// - we will try to preserve some significant
// bit-depths:
//   - NPX v1: 10-bit AP and LP streams
//   - NPX v2 beta: 14-bit
//   - NPX v2 final: 12-bit
```

jonnew, 2024-07-04:
@bparks13 Can you have a look at these scale factors when you have a moment and compare to those in the OE gui to ensure we are doing the same thing?

jsiegle, 2024-07-04:
One other thing to note is that 2.0 beta probes have a fixed gain of 80, versus 100 for the commercial probes.

bparks13, 2024-07-05:
@jonnew @glopesdev  I found a few discrepancies between the OE GUI implementation and the refactor.

## Neuropixels V1

- [OE GUI](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/cd4cbeb3feb5389facc634179bbd0319c6a54b15/Source/Probes/Neuropixels1.cpp#L422-L425)
  - 10-bit scaling (dividing by 1024)
  - Per channel gain correction
- [Onix-refactor](https://github.com/neurogears/onix-refactor/blob/963242afdee889cdd163b40b968e0630235aa088/OpenEphys.Onix/OpenEphys.Onix/NeuropixelsV1eDataFrame.cs#L27)
  - No bit scaling is applied that I can see
  - No gain correction applied, but ap/lfp gain are available in the method call

## Neuropixels V2 Beta

- [OE GUI](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/cd4cbeb3feb5389facc634179bbd0319c6a54b15/Source/Probes/Neuropixels2.cpp#L661-L662)
  - 14 bit scaling
  - Gain correction of 80
- Onix-refactor
  - [Bit scaling](https://github.com/neurogears/onix-refactor/blob/963242afdee889cdd163b40b968e0630235aa088/OpenEphys.Onix/OpenEphys.Onix/NeuropixelsV2eBetaDataFrame.cs#L45) is 14 bits
  - [Gain correction](https://github.com/neurogears/onix-refactor/blob/963242afdee889cdd163b40b968e0630235aa088/OpenEphys.Onix/OpenEphys.Onix/ConfigureNeuropixelsV2eBeta.cs#L92) is read from a file

## Neuropixels V2

- [OE GUI](https://github.com/open-ephys-plugins/neuropixels-pxi/blob/cd4cbeb3feb5389facc634179bbd0319c6a54b15/Source/Probes/Neuropixels2.cpp#L661-L662)
  - 12 bit scaling
  - Gain correction of 100
- Onix-refactor
  - [Bit scaling](https://github.com/neurogears/onix-refactor/blob/963242afdee889cdd163b40b968e0630235aa088/OpenEphys.Onix/OpenEphys.Onix/NeuropixelsV2eDataFrame.cs#L37) is set to 14 bits
  - [Gain correction](https://github.com/neurogears/onix-refactor/blob/963242afdee889cdd163b40b968e0630235aa088/OpenEphys.Onix/OpenEphys.Onix/ConfigureNeuropixelsV2e.cs#L74) is read from a file

jonnew, 2024-07-23:
Thanks Brandon. Yeah -- the gain correction is done on the hardware (fixed point multiplication) in the IMEC system. 

This is helpful, thank you.