open-ephys/bonsai-onix1 #129: Allow dynamic configuration of inner properties of hub devices

glopesdev, 2024-07-04:
Currently hub devices expose as public properties each of their internal devices for configuration. However, this makes it hard to dynamically assign properties to individual devices as inner properties cannot be directly externalized into the workflow (see https://github.com/bonsai-rx/bonsai/issues/1879).

The only option right now is to create copy constructors in each of the internal device configuration operators that allows constructing a new instance directly from values passed in the property mapping. For example, the following workflow would work to replace the Heartbeat configuration of the breakout:

[image]

<details>
<summary>
Workflow
</summary>

```
<?xml version="1.0" encoding="utf-8"?>
<WorkflowBuilder Version="2.8.1"
                 xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
                 xmlns:onix="clr-namespace:OpenEphys.Onix;assembly=OpenEphys.Onix"
                 xmlns:scr="clr-namespace:Bonsai.Scripting.Expressions;assembly=Bonsai.Scripting.Expressions"
                 xmlns="https://bonsai-rx.org/2018/workflow">
  <Workflow>
    <Nodes>
      <Expression xsi:type="Combinator">
        <Combinator xsi:type="onix:CreateContext">
          <onix:Driver>riffa</onix:Driver>
          <onix:Index>0</onix:Index>
        </Combinator>
      </Expression>
      <Expression xsi:type="Combinator">
        <Combinator xsi:type="IntProperty">
          <Value>0</Value>
        </Combinator>
      </Expression>
      <Expression xsi:type="scr:ExpressionTransform">
        <scr:Expression>new(
True as Enable,
it as BeatsPerSecond)</scr:Expression>
      </Expression>
      <Expression xsi:type="PropertyMapping">
        <PropertyMappings>
          <Property Name="Heartbeat" Selector="Enable,BeatsPerSecond" />
        </PropertyMappings>
      </Expression>
      <Expression xsi:type="Combinator">
        <Combinator xsi:type="onix:ConfigureBreakoutBoard">
          <onix:Name>BreakoutBoard</onix:Name>
          <onix:Heartbeat>
            <onix:DeviceName>BreakoutBoard/Heartbeat</onix:DeviceName>
            <onix:DeviceAddress>0</onix:DeviceAddress>
            <onix:Enable>true</onix:Enable>
            <onix:BeatsPerSecond>10</onix:BeatsPerSecond>
          </onix:Heartbeat>
          <onix:AnalogIO>
            <onix:DeviceName>BreakoutBoard/AnalogIO</onix:DeviceName>
            <onix:DeviceAddress>6</onix:DeviceAddress>
            <onix:Enable>true</onix:Enable>
            <onix:InputRange0>TenVolts</onix:InputRange0>
            <onix:InputRange1>TenVolts</onix:InputRange1>
            <onix:InputRange2>TenVolts</onix:InputRange2>
            <onix:InputRange3>TenVolts</onix:InputRange3>
            <onix:InputRange4>TenVolts</onix:InputRange4>
            <onix:InputRange5>TenVolts</onix:InputRange5>
            <onix:InputRange6>TenVolts</onix:InputRange6>
            <onix:InputRange7>TenVolts</onix:InputRange7>
            <onix:InputRange8>TenVolts</onix:InputRange8>
            <onix:InputRange9>TenVolts</onix:InputRange9>
            <onix:InputRange10>TenVolts</onix:InputRange10>
            <onix:InputRange11>TenVolts</onix:InputRange11>
            <onix:Direction0>Input</onix:Direction0>
            <onix:Direction1>Input</onix:Direction1>
            <onix:Direction2>Input</onix:Direction2>
            <onix:Direction3>Input</onix:Direction3>
            <onix:Direction4>Input</onix:Direction4>
            <onix:Direction5>Input</onix:Direction5>
            <onix:Direction6>Input</onix:Direction6>
            <onix:Direction7>Input</onix:Direction7>
            <onix:Direction8>Input</onix:Direction8>
            <onix:Direction9>Input</onix:Direction9>
            <onix:Direction10>Input</onix:Direction10>
            <onix:Direction11>Input</onix:Direction11>
          </onix:AnalogIO>
          <onix:DigitalIO>
            <onix:DeviceName>BreakoutBoard/DigitalIO</onix:DeviceName>
            <onix:DeviceAddress>7</onix:DeviceAddress>
            <onix:Enable>true</onix:Enable>
          </onix:DigitalIO>
          <onix:MemoryMonitor>
            <onix:DeviceName>BreakoutBoard/MemoryMonitor</onix:DeviceName>
            <onix:DeviceAddress>10</onix:DeviceAddress>
            <onix:Enable>false</onix:Enable>
            <onix:SampleFrequency>10</onix:SampleFrequency>
          </onix:MemoryMonitor>
        </Combinator>
      </Expression>
      <Expression xsi:type="Combinator">
        <Combinator xsi:type="onix:StartAcquisition">
          <onix:ReadSize>2048</onix:ReadSize>
          <onix:WriteSize>2048</onix:WriteSize>
        </Combinator>
      </Expression>
    </Nodes>
    <Edges>
      <Edge From="0" To="4" Label="Source1" />
      <Edge From="1" To="2" Label="Source1" />
      <Edge From="2" To="3" Label="Source1" />
      <Edge From="3" To="4" Label="Source2" />
      <Edge From="4" To="5" Label="Source1" />
    </Edges>
  </Workflow>
</WorkflowBuilder>
```
</details>

Assuming that the `ConfigureHeartbeat` class defines the following copy constructor:

```c#
public ConfigureHeartbeat(bool enable, uint beatsPerSecond)
    : base(typeof(Heartbeat))
{
    Enable = enable;
    BeatsPerSecond = beatsPerSecond;
}
```

bparks13, 2024-11-19:
@glopesdev after our discussion today I went back looking for this issue, is this in the same vein as the externalization for dynamic configuration or is this going to be a separate issue to tackle?

aacuevas, 2025-05-19:
There is a prototype in https://github.com/open-ephys/bonsai-netespropertymap-prototype 
The Bonsai team gave me some pointers on modifications. I've been working on them, but other priorities have been keeping me from this. The idea is, once I implement the proposed changes, to present them to the team and from there make a proper Bonsai.Core PR