open-ephys/bonsai-onix1 #136: Couldn't find visualizers for Quaternion data

ChucklesOnGitHub, 2024-07-19:
[image]

bparks13, 2024-07-19:
As far as I am aware, there is no direct visualizer for a Quaternion, or any of the other types with multiple axes (e.g., Acceleration, Gravity, etc.). What we can do is use a `RollingGraph` visualizer (found in the `Bonsai.Design.Visualizers` package) and connect it to the Quaternion output (see below).

[image]

The `RollingGraph` can then be visualized by selecting the **Bonsai.Design.Visualizers.RollingGraphVisualizer** option in **Show Visualizer**.

jonnew, 2024-07-20:
This was my main hesitation when moving away from just using Mats, which we were doing before. Its definitely correct to use these nice types but the lack of easy visualization is a big hit for users.

ChucklesOnGitHub, 2024-07-22:
Thanks. I had forgotten that difference between the old and the new library. This is actually already [addressed in the docs](https://cjsha.github.io/onix-bonsai-docs/articles/getting_started-visualizing_data.html). I've added the RollingGraph nodes to the example workflows. (You can follow work on example workflows here: https://github.com/cjsha/onix-bonsai-docs/issues/1).

jonnew, 2024-07-22:
@glopesdev Is there a way that we can give the rolling graph as default visualizer for Quaternion and Point data?

glopesdev, 2024-07-22:
@jonnew yes, but since this has to be done in the Design package, I would recommend doing this for the 0.2.0 package.

jonnew, 2024-08-13:
@glopesdev can you provide an example of how this has been done in another package? We might be able to take it from there.

glopesdev, 2024-08-21:
The quickest would be to reuse `TimeSeriesVisualizer` directly similar to [`ScalarVisualizer`](https://github.com/bonsai-rx/bonsai/blob/main/Bonsai.Vision.Design/ScalarVisualizer.cs).

Alternatively for more fine-grained customization you can make your own series plot similar to [`ScalarHistogramVisualizer`](https://github.com/bonsai-rx/bonsai/blob/main/Bonsai.Vision.Design/ScalarHistogramVisualizer.cs).

jonnew, 2024-08-21:
Thanks @glopesdev this worked great.

PR #238 (2024-08-21): Add visualizers for Quaternion and Vector3
- Both of these System.Numerics types are used data frames (Bno055DataFrame and TS4231V1PositionDataFrame)
- Fixes #136
closed 2024-08-21 by jonnew