open-ephys/bonsai-onix1 #140: ManagedFrame.FrameClock should be ManagedFrame.Clock

jonnew, 2024-07-22:
https://github.com/neurogears/onix-refactor/blob/260149bdfb441c8d49a606bbfd303728cbc42bd0/OpenEphys.Onix/OpenEphys.Onix/ManagedFrame.cs#L18C26-L18C27

Can I make this change and push to main?

glopesdev, 2024-07-22:
I would actually remove ManagedFrame entirely from the package. I don't think it is being used anywhere anymore and is no longer the recommended way to build a data frame type.

glopesdev, 2024-07-22:
Alternatively we could make it the common base type of all data frames, but a slight refactoring would be needed, I can bring up a proposal in a PR for tomorrow.

jonnew, 2024-07-23:
Its used by the heartbeat. The reason I noticed is because I was looking for a place to document Clock and HubClock which are repeated everywhere in the library.

jonnew, 2024-07-23:
Re: refactor -- I don't think we need to make a common base class if there is not a good reason, its a minor thing I noticed during documentation.

glopesdev, 2024-07-24:
@jonnew I think there are good reasons to having a common base class for this, as it makes it easier to consistently change the naming conventions, introduce new common properties we may not be exposing right now (e.g. `DeviceAddress`), and also facilitates documentation efforts since these properties only have to be documented in a single place.

I have opened a PR (#147) with a proposal for this which also refactors the heartbeat data and gets rid of `ManagedFrame`.

PR #147 (2024-07-25): Refactor common data frame properties into abstract base type
All ONI frames have a common structure which is used by all device data streams. This structure is currently replicated in the various `*DataFrame` types defined in the library. This generates redundancy and complicates both documentation and refactoring, since changes to this common signature are not tied together.

This PR introduces two new abstract base classes: `DataFrame` and `BufferedDataFrame` to help provide common properties for classes exposing data from either a single frame, or data buffered from multiple frames, respec [... cut]
closed 2024-07-25 by glopesdev