open-ephys/bonsai-onix1 #160: Class names DigitalIO, AnalogIO, and related are too generic

jonnew, 2024-07-25:
I think that the classes  `AnalogIO` and `DigtialIO`, along with all related classes are too generically named. These devices are pretty idiosyncratic to their use with the FMC host/breakout board combination. Obvious reasons why:

- Analog IO has 12 channels, which is frankly a weird number
- Digital IO contains button states that only make sense in context of breakout board

Even at the level of liboni, `DigitalIO` is referred to as `ONIX_BREAKDIG1R3` indicating its localization to the breakout board. Similarly, `AnalogIO` is `ONIX_FMCANALOG1R3` because the sampling circuit is in fact localized the FMC module in the computer. In this case, I think `BreakoutAnalogIO` is still fine because this is where the user will intact with the analog system.

glopesdev, 2024-07-25:
@jonnew I agree, but don't forget we are renaming the entire package as `OpenEphys.Onix.FmcHost` which is why last time we agreed to just leave the names as they are for now.

glopesdev, 2024-07-25:
To clarify, this means in the toolbox people will see `AnalogIO (OpenEphys.Onix.FmcHost)`

PR #164 (2024-07-26): Rename AnalogIO and DigitalIO to refer to breakout
To align with ONI device descriptors and physical hardware usage, we are renaming these general device types to reflect their physical instantiation in the FMC host breakout board.

Fixes #160
closed 2024-07-26 by glopesdev