open-ephys/bonsai-onix1 #173: Missing documentation for package operators and public classes

glopesdev, 2024-07-27:
While reorganizing the project for packaging, I noticed the following operators and types remain undocumented:

- [`ContextTask`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/ContextTask.cs)
- [`CreateContext`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/CreateContext.cs)
- [`ConfigureHarpSyncInput`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/ConfigureHarpSyncInput.cs)
- [`ConfigureLoadTester`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/ConfigureLoadTester.cs)
- [`DeviceFactory`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/DeviceFactory.cs)
- [`HarpSyncInputData`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/HarpSyncInputData.cs)
- [`HarpSyncInputDataFrame`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/HarpSyncInputDataFrame.cs)
- [`HubDeviceFactory`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/HubDeviceFactory.cs)
- [`PassthroughState`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/PassthroughState.cs)
- [`StartAcquisition`](https://github.com/neurogears/onix-refactor/blob/6dc9e26d38a48b9a0f5c13ffb0a8966236ba2144/OpenEphys.Onix/OpenEphys.Onix/StartAcquisition.cs)

We could potentially internalize some of these as implementation-specific classes, which would simplify both the API and docs effort at this point, but the operators will need to be included.

jonnew, 2024-07-27:
> Assigned #173 <https://github.com/neurogears/onix-refactor/issues/173> to
> @jonnew <https://github.com/jonnew>.

Yep on my to-do tomorrow. Thanks for pulling out the missing ones.

glopesdev, 2024-07-28:
@jonnew great, let me know if you want to merge #172 to main beforehand so you have a few less classes to worry about.

jonnew, 2024-07-28:
That would be helpful. Im going to fork from 172 for my working copy

glopesdev, 2024-07-28:
Great, no need to fork, I have merged it now.

PR #177 (2024-07-30): Add missing documentation and description strings for all public classes
Following #175 this PR completes the technical docs by adding missing documentation comments to all classes and members in the public API surface.

Description attributes were also added to all top-level operator classes to allow displaying summary strings in the toolbox.

Fixes #173