open-ephys/bonsai-onix1 #176: Possible FmcLinkController issue with duplicate names

bparks13, 2024-07-29:
While working with Jakob, I merged changes made in `main` to `issue-88` (NPX 2.0e GUI branch) so he could continue testing with the latest changes.

However, he ran into some issues with the `ConfigureNeuropixelsV2eHeadstage` node, where it would complain that a device with the same name had already been configured (see image below).
[image]

If he changed the name of the headstage to "npx", it would configure correctly but then throw a buffer size error, which would go away if he increased the buffer size.

I know that there were some changes made to the `FmcLinkController` class, I am going to dig into it more tomorrow to see if I can find the issue, but wanted to document this process in case someone else can help resolve it.

bparks13, 2024-07-29:
@jonnew @glopesdev any ideas on what might be the issue?

jonnew, 2024-08-13:
@jvoigts Does this issue arise with your workflows that are using "Retry"? If so, its not so much a specific code issue but the fact that a Context is hanging around when it should have been disposed.

jvoigts, 2024-08-13:
Looks like i dont get that with start>retry.
I have definitely seen this error though, without any 'retry' craziness.

Not sure if relevant, but one way can easily recreate the _inverse_ error (No device with that name has been configured) is to add a delay before the headstage config node. In this case the data node throws this error. Maybe that is expected because the data node looks for the device immediately. This made me naively think the data nodes should be connected to the config nodes to ensure this order is correct?

closed 2024-10-28 by jonnew