open-ephys/bonsai-onix1 #192: Rapid restarting of a NeuropixelsV2e workflow gives 'attempt to trigger already triggered operation' error.

jvoigts, 2024-08-03:
This can occur either when using the restart button in bonsai or stopping and starting rapidly (within about a second maybe).
I'm seeing this with two V2e neuropixels, nothing else in the workflow. I have not properly tried different workflows etc. yet, so I have no info on whether this is specific to this configuration.

jonnew, 2024-08-04:
If a `NeuropixelsV2eBno055Data` is feed with a high-frequency sequence, the user will very likely be presented with an "Attempt to trigger and already triggered operation" exception. Two issues:

- This error string is really uninformative. What it means is that there is an ongoing read or write to a device register when another is attempted. Issue to improve these strings here: https://github.com/open-ephys/liboni/issues/26
- The presence of the error reveals a threading issue that is probably always present in this library but has a high probability of occurring when `NeuropixelsV2eBno055Data` (and nodes that behave like it) are feed with a high-frequency sequence. We found this issue by feeding the ``NeuropixelsV2eBno055Data` with the output of `NeuropixelsV2eBnoData` which is produced at ~30 kHz. What seems to be happening is that ``NeuropixelsV2eBno055Data` is attempting to access the configuration channel to read a BNO value while its being used to for the disposal processes that occurring during a workflow restart. Obviously this is not an advisable topology but its valid so we need to fix.

Side note: I would attach a workflow but I dont know how because of githubs file type filters.

glopesdev, 2024-08-04:
@jonnew I think the issue that @jvoigts raised is a separate one from this issue, since it is a known issue with asynchronous dispose (i.e. the previous context may be disposing while a new one is being created).

I think the high frequency polling issue is different even though they output the same low-level oni esception, so I would open two separate issues until we conclude definitively that they are the same.

jonnew, 2024-08-04:
I dont think so -- we worked together to figure out what would fix the original issue he raised and this was the cause

glopesdev, 2024-08-05:
@jonnew ok, just to confirm since I am realizing your original comment did not fully clarify, when is the error exactly thrown? Is it still when the user starts and stops the workflow? Or an error during polling at runtime?

If the former then I agree it might be the same error.

jonnew, 2024-08-05:
I should have made I clearer (it's the 2nd to last sentence of the second bullet) that this occurs during a workflow restart. Without a high frequency input to the `NeuropixelsV2eBno055Data` node the workflow can be restarted without issue.