open-ephys/bonsai-onix1 #198: Reformat remarks for boolean properties to take advantage of XML definition lists

cjsha, 2024-08-08:
Refer to [this issue in the docs repo](https://github.com/open-ephys/onix1-bonsai-docs/issues/10#issue-2455973658) to understand what this issue proposes to accomplish. 

Supposing that we all agree with this issue's goal, the remarks for all boolean properties should be reformatted into [XML definition lists](https://learn.microsoft.com/en-us/dotnet/csharp/language-reference/xmldoc/recommended-tags#list). 

The commit that closes this issue should only be merged into main after the docs template has custom html/css for XML definition lists (basically only after [the aforementioned issue](https://github.com/open-ephys/onix1-bonsai-docs/issues/10#issue-2455973658) closes).

jonnew, 2024-08-13:
This is a nice idea. Can you propose and exact example syntax for the XML comment on a boolean property to get what you need?

cjsha, 2024-08-20:
```
        /// <remarks>
        /// <list type="table">
        /// <item>
        /// <term> True </term> 
        /// <description> A <see cref="Bno055Data"/> instance that shares a <c>DeviceName</c> will produce data </description>
        /// </item>
        /// <item>
        /// <term> False </term> 
        /// <description> A <see cref="Bno055Data"/> instance that shares a <c>DeviceName</c> won't produce data. </description>
        /// </item>
        /// </list>
        /// </remarks>
```

would yield 
[image]
`DeviceName` contains a space in the screenshot but it shouldn't. We don't use them at all, but those `<c></c>` tags are valid XML comment tags as well.

jonnew, 2024-08-20:
I think the result is more confusing than just writing the sentence to be
honest

cjsha, 2024-08-20:
If you look again at [the original issue in the docs repo](https://github.com/open-ephys/onix1-bonsai-docs/issues/10#issue-2455973658), do you feel the same way?

jonnew, 2024-08-20:
No, i guess not. It does seem clearer with with bullets.

However, both of these docs have a problem which is that the Configuration node cannot say what data node to use because they are linked only through a device type. There could be different data nodes for the same configuration node. And with e.g. the `TS4231xx` stuff there are. So, this has nothing to do with the formatting of the boolean anymore but is a problem in both cases.

cjsha, 2024-08-20:
ok I opened a separate issue for that. 

If you think this is something we want to do (reformatting remarks), feel free to assign me and set a milestone.

closed 2024-08-26 by jonnew