open-ephys/bonsai-onix1 #201: Use inheritdoc or include tags in xml comments

cjsha, 2024-08-09:
Should commonly repeated text should be inherited xml comments? DRY principle makes sense to me here.

[inheritdoc](https://learn.microsoft.com/en-us/dotnet/csharp/language-reference/xmldoc/recommended-tags#inheritdoc) and [include](https://learn.microsoft.com/en-us/dotnet/csharp/language-reference/xmldoc/recommended-tags#include) are standard XML comments and supported by docfx.

For example, the same text is used in these places:

https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2eBeta.cs#L53-L56

https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2eBeta.cs#L72-L75

https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV1e.cs#L92-L95

https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2e.cs#L43-L46

https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2e.cs#L62-L65

bparks13, 2024-08-09:
I definitely agree in principle, but the hard part is that we can't use `<inheritdoc>` here because these are all unique classes (maybe there is an argument to have a base class for Neuropixels, but I'm not sure that is worth it). 

So then the question is how can we leverage the `<include>` tags efficiently? I don't like the idea of choosing one of these files semi-arbitrarily to write the documentation string, and then linking the path to that string in all of the other files. That feels like a problem if we repeat this pattern anywhere, because now we need to maintain the linkages to an arbitrary end-point. 

This could be a good argument for maintaining external files that hold documentation that all variables then point to, but I am somewhat opposed to that because I don't like separating the documentation from the code. From the standpoint of the documentation website it won't make any difference though, so I could be convinced.

cjsha, 2024-08-13:
I think Brandon brings up some good points. It seems like we are leaning towards not doing this

However, before closing this completely, @jonnew mentioned we should to figure out if we can include/inherit just a member's remarks. I wouldn't try doing this with the <inheritdoc> tag. I would use the [<include>](https://learn.microsoft.com/en-us/dotnet/csharp/language-reference/xmldoc/recommended-tags#include) tag. From my interpretation of the example on the linked page, I think we can transfer just the remarks from a single source of truth that gets proliferated to members that have duplicate remarks.

cjsha, 2024-08-13:
If we want, I'll make a branch so we can see what this looks like. I can look for other repeated text so we can see what it looks like for multiple transclusions. It's not super concrete in my mind - I'm not sure how clunky it will end up being

jonnew, 2024-08-26:
- Really what we want is `<include_remarks>` to take the `<remarks>` section for some other XML comment but not the whole XML comment. 
- I am not a fan of using `<include>`  and separating code and XML comments. I want the comments to be a reflection of the current state of the code and the best way to enforce that is for them to be in the source.

closed 2024-08-26 by jonnew