open-ephys/bonsai-onix1 #202: ProbeConfigurationA & ProbeConfigurationB properties lack documentation

cjsha, 2024-08-09:
I assume that documentation for these NeuropixelsV2eBeta and NeuropixelsV2e properties  are postpone until the configuration GUIs are available for the 0.2.0 release. Either way, I want to note that info is missing in the interim, at least as a reminder.

[image]

If the probe configuration GUI is sufficiently intuitive, it might suffice to put some instructions on how to use the GUIs in the remarks which get populated here in this table? Not sure how much instruction will be necessary here.

cjsha, 2024-08-12:
The locations of the potential \<remarks\> comments to would be here:
https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2eBeta.cs#L43-L48
https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2eBeta.cs#L62-L67

cjsha, 2024-08-12:
and also here:
https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2e.cs#L33-L38
https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/ConfigureNeuropixelsV2e.cs#L52-L57

jonnew, 2024-08-13:
I think that all we have to do here is to add a <remark> section that states a sentence or two about how these properties are generally configured via a GUI.

bparks13, 2024-08-14:
`NeuropixelsV2e` probe configuration variables are updated in #214.

When GUIs are added for `NeuropixelsV2eBeta`, the probe configuration variables will be updated accordingly.

bparks13, 2024-08-20:
Moving this to 0.3 since we are not going to release `NeuropixelsV2eBeta` GUIs until then.

PR #269 (2024-09-04): Add NeuropixelsV2eBeta GUIs
This PR adds GUIs for the `NeuropixelsV2eBeta` headstage. The added dialogs call the `NeuropixelsV2e` GUIs, taking advantage of the common interface between `ConfigureNeuropixelsV2e` and `ConfigureNeuropixelsV2eBeta`. When the underlying type is a `*Beta`, all relevant text strings are updated to reflect this.

Closes #227
Closes #268 
Fixes #202
closed 2024-09-04 by jonnew