open-ephys/bonsai-onix1 #207: Description connecting <xyz>Data operators and configure<xyzHeadstage> operators

cjsha, 2024-08-13:
[image]

The part underlined by red is repeated throughout the docs (besides [BreakoutDigitalInput](https://open-ephys.github.io/onix1-bonsai-docs/api/OpenEphys.Onix1.BreakoutDigitalInput.html), also [Bno055Data](https://open-ephys.github.io/onix1-bonsai-docs/api/OpenEphys.Onix1.Bno055Data.html) for example). What does it mean to be "linked"? I'm not sure a naive user would know. I would replace it with a more concrete description such as:

> The BreakoutDigitalInput ``DeviceName`` property must correspond to the ``DeviceName`` property of a ``ConfigureBreakoutBoard`` operator to produce data.

or something like that

The corresponding code for this specific example is:

https://github.com/open-ephys/onix-bonsai-onix1/blob/7ac700fb8a696d4d9f8d97c9d1b85765b9951ca6/OpenEphys.Onix1/BreakoutDigitalInput.cs#L12-L15

jonnew, 2024-08-26:
Related to #229

jonnew, 2024-09-03:
There are two issues that need to be addressed here

1. The complete process of setting up a configuration chain and then setting up 'receiver' operators to capture data from the devices defined in the configuration needs to be described in high level terms in the getting started guide. There should be 2 major pages to do this: Configuration and Acquisition. The first focuses on describing and defining configuration chain. The second focuses on linking Data operators to this chain, running the workflow, and visualizing data.
2. The XML description should not just say "link" but should reference and describe the `DeviceName` property that performs the "linking" ( e.g. linked by selecting the corresponding `DeviceName` property value).

PR #288 (2024-09-06): Document how DeviceName links configuration and data IO operators
- Added remarks to all device configuraiton and data IO operators indicating that they can be linked through the use of a shared DeviceName
- Fixes #207
closed 2024-09-06 by jonnew