open-ephys/bonsai-onix1 #211: Headstage64 ElectricalStimulator & OpticalStimulator Remarks require more information

cjsha, 2024-08-13:
There should be more instructions on how to use the ``ElectricalStimulator`` and ``OpticalStimulator`` properties on the ConfigureHeadstage64. I think it would be proper to include more info in their remarks? in addition to other pages in the docs.

[image]

Also, I know these properties don't necessarily belong in the Misc subheader, this is its own [issue that belongs to the docs repo](https://github.com/open-ephys/onix1-bonsai-docs/issues/11).

jonnew, 2024-09-05:
I think this might be handled by #288 because now these Configuration operatators will have a link the the trigger operators in their remarks, which provide more information on use. Beyond that, I think its best left to the headstage-64 user guide article, unless I'm not understanding something.

The misc classification issue was fixed in #226.

If you disagree, keep this open and let me know. Otherwise, I'm voting to close.

cjsha, 2024-09-06:
I actually don't see where these have links to the trigger operators in their remarks. They don't have remarks at all in that issue-207 branch which is the one being merged in #288

https://github.com/open-ephys/onix-bonsai-onix1/blob/e020b3313bc3825954eac22c00dec716ce1d6f40/OpenEphys.Onix1/ConfigureHeadstage64.cs#L81-L87
https://github.com/open-ephys/onix-bonsai-onix1/blob/e020b3313bc3825954eac22c00dec716ce1d6f40/OpenEphys.Onix1/ConfigureHeadstage64.cs#L89-L95

but in any case, the original post for this issue is old. This is how this page looks now: 
https://open-ephys.github.io/onix1-bonsai-docs/api/OpenEphys.Onix1.ConfigureHeadstage64.html
[image]

That's generic text that the website serves when a device doesn't have public properties. I think it's insufficient - people won't know that they need to connect a trigger operator to get the stimulator functionality going. If you write remarks where I github perma-linked above, I can probably put those remarks there.

bparks13, 2024-09-06:
@cjsha it looks like the comments linking the configure node to the trigger operator exist in the device nodes. Would it be sufficient to use `<inheritdoc>` from the configure device node in the configure headstage node? I remember seeing something about a hierarchical inheritance, so we can keep the summary written in the headstage node, but pull in the remarks to then display the correct information here.

jonnew, 2024-09-06:
Im going to re-assign this one to @bparks13 to explore these ideas.

PR #292 (2024-09-06): Explicitly link `Trigger` operators to their constituent `Configure` operator
Currently there is no indication of how to link a `Trigger` operator to the `Configure` operator. By adding remarks in this PR, this linking is now explicitly stated.

Fixes #211
closed 2024-09-06 by cjsha