open-ephys/bonsai-onix1 #218: Add Title field to the project files

jonnew, 2024-08-15:
Add a `<title>` field to the main and design packages so that their Nuget display name more user friendly

[image]

However, from @glopesdev 

> just a warning that NuGet does not respect the Title tag anymore and now just shows the unique package ID (I'm guessing to prevent impersonation). Not sure how their client API and the package manager will evolve in the future but this feature might not be guaranteed to remain in the future

bparks13, 2024-08-15:
What would be the ideal display name? Following the pattern in other `<Title>` tags shown here, we could name the base library "Bonsai Onix1 - Open Ephys" or some variation of that.

Similarly, for the Design library we can call it "Bonsai Onix1 Design Library - Open Ephys" or something like that.

jonnew, 2024-08-16:
I like those two options. I also like applying that schema to the rest of the packages.

glopesdev, 2024-08-16:
I don't see how adding the word "Bonsai" in this case adds more information since the `Title` is already ignored by NuGet in general, and only used inside Bonsai package manager where people already expect all packages to be Bonsai packages.

Furthermore we are redesigning the package manager details page to address issues raised by @cjsha, and to increase consistency with NuGet we are emphasising package IDs over titles.

I can show you what it looks like next time we meet and we can discuss the pros and cons.

cjsha, 2024-08-16:
[This](https://github.com/bonsai-rx/bonsai/issues/1965) is what glopesdev is referring to in the last comment btw.

If we're choosing between those two titles that bparks mentioned, I like the first one more basically because of glopes mentioned. 

With that said, we refer to the package id ``OpenEphys.Onix1`` all over the docs. We give people instructions what to look up and select in the package manager. And the package id shows up in Bonsai's module search bar. All this is to say I'm kinda ok with ``OpenEphys.Onix1`` as the thing that people look for in package manager. The ``Title`` can be anything, but the package id is the package id, at least as far as I understand it. Imo, it's less ambiguous with little cost to readability. 

[image]
[image]

jonnew, 2024-08-16:
The reason why the title might want to include `Bonsai` is because this is a bonsai package:

- This information is no longer in our package name or namespace because we are trying to follow best practices for naming that
- Our package is in the  `All` package source, which is filled with nuget packages that have nothing to do with bonsai. 
- If our package showed in the `Bonsai` feed, or was properly filtered using tags (maybe this is coming) then i would agree with all this.

jonnew, 2024-08-20:
It looks like `Title` is basically be deprecated

1. Focus on a standard description that starts with `Bonsai library for...` if its a Bonsai package
2. Set the nuget package type (https://learn.microsoft.com/en-us/nuget/create-packages/set-package-type?tabs=dotnet). To do this, in the `csproj` <ProjectGroup>:  `<PackageType>Dependency;BonsaiLibrary</PackageType>`

closed 2024-08-22 by jonnew