open-ephys/bonsai-onix1 #229: There could be different data nodes for the same configuration node. And with e.g. the `TS4231xx` stuff there are. So, this has nothing to do with the formatting of the boolean anymore but is a problem in both cases.

cjsha, 2024-08-20:
However, both of these docs have a problem which is that the Configuration node cannot say what data node to use because they are linked only through a device type. There could be different data nodes for the same configuration node. And with e.g. the `TS4231xx` stuff there are. So, this has nothing to do with the formatting of the boolean anymore but is a problem in both cases.

_Originally posted by @jonnew in https://github.com/open-ephys/onix-bonsai-onix1/issues/198#issuecomment-2299477489_

This seems to be a pattern repeated through the XML comments. 

I see two options for resolving what Jon mentioned:
1. Remove those links in all `configure<Whatever>` operators
2. Remove those links in `configure<Whatever>` operators that can correspond to multiple `<whatever>Data` operators

bparks13, 2024-08-21:
Is it too much effort to add links to the multiple different `<whatever>Data` nodes? That way people can still navigate to the correct one that they have, without having to manually search for it.

cjsha, 2024-08-21:
Oh yeah. That's a valid 3rd option. I suspect that it's best to put that info somewhere in the docs repo instead of here though.

closed 2024-08-26 by jonnew