open-ephys/bonsai-onix1 #234: Add linkchecker for xml comments in gh actions triggered by PRs

cjsha, 2024-08-21:
To avoid issues like #233, it would be nice to have some kind of linkchecker.

cjsha, 2024-08-21:
That makes sense to me. Do you think there's a way that isn't clunky to build the docs site and check its links in this repo?

bparks13, 2024-08-21:
Not easily, no. To build the docs from this repo you would have to go to the docs repo and 1) create a new branch, 2) update the submodule, and push that commit, 3) open a PR to trigger the Github actions, 4) monitor the status of the actions, 5) only pass the checks in this repo if all the checks in the docs repo pass. 

This might be possible if we have the documentation in the same repo, so that instead of using a submodule it is all in one place, but I think that we decided against that for a number of reasons. As of right now, I think the only way to check if a link is broken is to test it from the docs repo.

jonnew, 2024-08-26:
The resolution here is to open an issue pointing to the reference in the XML comments that indicates a broken link and we can fix it from there.

closed 2024-08-26 by jonnew