open-ephys/bonsai-onix1 #247: Proposal: DataFrameWriter

jonnew, 2024-08-22:
Sometimes we use `enum`s as members of data frames:

https://github.com/open-ephys/onix-bonsai-onix1/blob/e5e5a97a956b60e4bd04fa9915c5774278579f90/OpenEphys.Onix1/BreakoutDigitalInputDataFrame.cs#L26

These can be nice because they can be used with `HasFlag` to determine if they contain some value (e.g. a pin is high or button was pressed in this case). However, in the context of Bonsai's existing infrastructure, they are basically dead-on-arrival when it comes to saving data. We have two options that I can see, currently:

1. Use `CsvWriter`: This option is not good. We can look some `;`-delimited example data produced by the breakout board to see why:

```
...
4410751;Pin0, Pin1, Pin2, Pin3, Pin4, Pin5;PortBOn, PortAOn
4410813;Pin0, Pin1, Pin2, Pin3, Pin4, Pin5, Pin6, Pin7;PortBOn, PortAOn
4416751;Pin0, Pin1, Pin2, Pin3, Pin4, Pin5;PortBOn, PortAOn
4416813;Pin0, Pin1, Pin2, Pin3, Pin4, Pin5, Pin6, Pin7;PortBOn, PortAOn
...
```
- The first element is a `ulong`
- The second element is all the bit flags within the `BreakoutDigitalPortState` that are high. In reality, this variable sized, human readable array with the word "Pin" repeated over and over again is encoding a byte. Not smart. Not only extremely inefficient but a complete nightmare to even parse in a reasonable way.
- The last element is the same story for the state of buttons and switches on the breakout.

2. We could split out each element of the data frame and use a `MatrixWriter`. The caveat is that `MatrixWriter` requires an array of privative types. So for each element of the data fram we must convert the enum to its underlying type using an `ExpressionTransform`: `byte(it)`, then `BufferCount` the byte into a single  or multielement Enumerable, then convert to an array, and then use matrix writer. Not good. There is some hope in this issue: https://github.com/bonsai-rx/bonsai/issues/1578. In that case we could just use `BinaryWriter` (or whatever it ends up being called) directly on the enums, but we don't know the timeline. 

The real issue here is probably bigger than just this: I think we need some type of generic `DataFrameWriter` sink that can go right after any of the `XXXData` nodes to serialize their streams to disk in a reasonable way, rather than having to split out elements before saving. I know that @glopesdev had something for serializing `oni.frame` data to individual files based on the device address, but this is too low level IMO. The `XXXData` nodes often perform a good deal of reorganization and processing that would need to be redone offline.

Here is a non-functional idea. This code compiles, but for reasons I don't yet understand, Bonsai will not load the operator. We may just need to make a concrete `DataFrameWriter` for every correspond `DataFrame` type instead. Maybe we could banish this atrocity to its own `OpenEphys.Onix1.Utils` package.

``` csharp
public abstract class DataFrame
{
    internal DataFrame(ulong clock)
    {
        Clock = clock;
    }

    internal DataFrame(ulong clock, ulong hubClock)
        : this(clock)
    {
        HubClock = hubClock;
    }

    public ulong Clock { get; }

    public ulong HubClock { get; internal set; }

    internal virtual void Write(BinaryWriter bw)
    {
        bw.Write(Clock);
        bw.Write(HubClock);
    }
}

  [Description("Records frame.")]
  public class DataFrameWriter<Tsource> : Sink<Tsource> where Tsource : DataFrame
  {
        const string fn = "./test.raw";

        public override IObservable<Tsource> Process(IObservable<Tsource> source) 
        {
            return Observable.Using(
                () => new BinaryWriter(new FileStream(fn, FileMode.Create)),
                bw => source.Do(f =>
                {
                    f.Write(bw);
                }).Finally(bw.Close));
        }
  }
```

aacuevas, 2025-05-19:
So first of all, it needs to actually write stuff to disk. This is a proof of concept that uses reflection and expression techniques to extract properties from frames and routes to different methods that convert most types into a byte[] that is then fed to a Write method, that currently just prints the array on the console. So what needs to be done is:
- Implement the actual disk writing (it needs buffering, a write thread, etc...) Alternatively, we could just make this convert classes to a byte array and use Bonsai's MatrixWriter to do the actual writing
- Implement conversion methods for other needed classes that don't fit into the general approach (e.g.: pImage)

Optionally, I don't know if we should add some kind of header to the files (If we take care of writing ourselves) to describe the format of the file.

jonnew, 2025-05-20:
OK, so after reviewing the code and talking with you on Discord I think we've reached the following conclusions

1. The strategy seems good.
2. We should rely on `MatrixWriter` to do the actually writing since its already battle tested, multi-threaded, and exposes a commonly used user interface for saving files. Additionally, we did some tests to make sure we can feed it observables containing arrays of different length and it worked fine. The final solution can be implemented as an higher order operator, implemented in Bonsai and packaged in the Onix1 library,  that combines the `FrameWriter` and `MatrixWriter`, exposes the `MatrixWriter` properties, and forwards the `DataFrame` to make the operator into a proper sink.
3. We will need to do some optimization: namely, there should be one byte[] array per incoming frame. We can rely on the fact that both `DataFrame` and `BufferedDataFrame` are, by definition, fixed size (even though they use dynamically sized types like `CV.Mat` to store data). This means that we must limit `FrameWriter` to operating on these types. 
4. We must come up with a standard, easy to version, and hopefully already existing, header format.

glopesdev, 2025-05-20:
@jonnew an alternative to `MatrixWriter` might be to simply target directly its parent class `StreamSink` which provides the whole infrastructure for multi-threaded writing to streams.

Have a look at the implementation of `MatrixWriter` itself and you will see that for the case `byte[]` it is actually practically trivial: https://github.com/bonsai-rx/bonsai/blob/main/Bonsai.Dsp/MatrixWriter.cs/

This way you would save having to allocate any intermediate arrays and simply run your serialization code straight to a `BinaryWriter`.

As for the header format, this might be a challenge, but why not use the ONIX format itself? Doesn't it already contain everything you need? Even if you don't save the "raw" data in itself, you could still define these new "processed" streams as ONI frames, right?

That would be nice since then you can simply have a single loader in Python which could handle raw data packets and logging packets interchangeably with no change required.

aacuevas, 2025-05-20:
We need a different format. A ONI frame is not always the same as a `DataFrame` object, since some frames do require some post-processing that is done in software (an example of this is miniscope frames. An ONI frame is a single line, but the `DataFrame` contains a full `pImage`.

There is another related consideration. We are using `Type.GetProperties()` to get a list of the properties on each frame object, so we don't have to manually create a list for each class. However, according to the documentation, and at least on the .NET Framework version that Bonsai targets, the order of the returned properties is not guaranteed. @glopesdev do you know of a way around that?

glopesdev, 2025-05-20:
> We need a different format. A ONI frame is not always the same as a DataFrame object, since some frames do require some post-processing that is done in software

I am not saying to use the raw frames themselves, but to encode the post-processed data into an ONI frame (with a new layout) but based on the common ONI header.

It is surprisingly not so easy to find well-supported efficient formats for binary file storage that support high-speed streaming to disk. Alternatively you can store flat binary with a JSON metadata file on the side.

I would be very wary of defining a different format as it will keep multiplying the maintenance burden and reduce the tooling surface. I would say either reusing ONI or a strong existing standard should be the main options considered.

(btw, if you do find a good existing standard for this let me know and I would probably be happy to make a generic binary writer for it to include in the standard library. protobuf looks like a possible contender for this but still not ideal)

> the order of the returned properties is not guaranteed

There is a way to guarantee which we use on the CsvWriter and other generic similar writers in Bonsai which is to sort by a reflection attribute, I don't remember off the top of my head but it will be in the source code.

aacuevas, 2025-05-21:
> I am not saying to use the raw frames themselves, but to encode the post-processed data into an ONI frame (with a new layout) but based on the common ONI header.

To be honest, I am not sure what are you referring to. 

In any case, out idea was to serialize frames into byte arrays and just store one frame after the other. One `FrameWriter` per device, thus each file being just a collection of frames, all with the same structure.

What we need is to store the schema, so any loader can know "Each frame has an int64 and then a matrix of 256x32 uint16", for example. ONI does not have any way to store this information, beyond the datasheets. Nor have we any proper tool to load data yet. 

This could very well be in a json besides the file, instead of a fixed header. But even with a json, we'd still prefer a schema to store the binary organization that already exists. 

> There is a way to guarantee this which we use on the CsvWriter and other generic similar writers in Bonsai which is to sort by a reflection attribute, I don't remember off the top of my head but it will be in the source code.

From what I've seen, you're sorting by `MemberInfo.MetadataToken`. Is this always guaranteed to follow the order in which members are defined inside a class in the code? I could not find any info on that.

glopesdev, 2025-05-21:
> ONI does not have any way to store this information, beyond the datasheets.

This is what could be improved, and it would potentially help interfacing with the devices too. The mechanism would be similar to the device schemas we use in Harp, where given a unique device ID, there is a unique schema describing the frame layout of that device.

If you had this, you could automatically generate documentation, the internal frame payload structs, python code for opening serialized binary frames, etc, exactly as we do for Harp.

Then defining a "processed frame" layout would be simply a matter of defining a schema for the data layout and assign it a unique ID. The schema can live next to the data and be used by a data access API to open everything as we do with [harp-python](https://github.com/harp-tech/harp-python).

I do see how the work to do this for ONI is not yet done, but at least having a formal description of data sheets as a JSON schema would be a big step towards having machine readable documentation and be helpful in laying out ONI as a standard regardless of whether you end up using it for data storage or not.

If this approach sounds interesting I am happy to meet to explain what we did for Harp and you can then take whatever bits are useful.

Otherwise, I feel something like protobuf might suit your needs. You always need to define a schema anyway, so everything I said above would still apply.

> Is this always guaranteed to follow the order in which members are defined inside a class in the code?

It was never officially guaranteed in documentation, but in practice all the implementations guarantee this at least up to .NET 7.

This *could* theoretically change in the future but [from .NET 7 onwards the reflection implementation is guaranteed to return properties in declaration order](https://github.com/dotnet/runtime/issues/46272) exactly so this is never a problem in the future.

To summarise, .net < 7 is guaranteed by implementation to work, and .net >= 7 guarantees it by reflection, dropping the need to sort, so a way to future proof the code should be something like the below pseudocode:

```c#
#if NET7_OR_GREATER
GetProperties()
#else
GetProperties().OrderBy(metadataToken)
#endif
```

glopesdev, 2025-05-21:
@aacuevas actually I just found out about a really nice binary serialization format which has all the right caracteristics: https://avro.apache.org/docs/

Will try and make a mock for generalised data serialization and if it works we could consider simply using Avro?

aacuevas, 2025-05-21:
> Then defining a "processed frame" layout would be simply a matter of defining a schema for the data layout and assign it a unique ID. The schema can live next to the data and be used by a data access API to open everything as we do with [harp-python](https://github.com/harp-tech/harp-python).

This is exactly our intention, so yes then. A per-device schema and a serialized layout that could easily be deserialized by knowing the schema. We just don't want to reinvent the wheel with Yet Another Schema Format, of which I'm sure there are hundreds by now.

> @aacuevas actually I just found out about a really nice binary serialization format which has all the right caracteristics: https://avro.apache.org/docs/

Nice find! It does seem to have all the right characteristics indeed. Let us first look at it properly and discuss, but seems promising. Thanks!

> It was never officially guaranteed in documentation, but in practice all the implementations guarantee this at least up to .NET 7.

Perfect, thanks! We'll use `MetadataToken` to sort the properties then.