open-ephys/bonsai-onix1 #248: Ensure voltage algorithms are tuned for OEPs standard tethers

jonnew, 2024-08-23:
The headstage specific auto port voltage tuning algorithm, e.g.:

https://github.com/open-ephys/onix-bonsai-onix1/blob/67f250de87cb825ee0f5091e4e3202032db79512/OpenEphys.Onix1/ConfigureHeadstage64.cs#L143

were not created using OEPS-supplied standard tether lengths in all cases. We need to ensure that the parameters of the algorithm are tuned to function well with the tethers provided by OEPS.

jonnew, 2024-08-26:
@filcarv What are the standard tether types and lengths that are used witch each ONIX headstage that OEPS is distributing?
- Npx 1.0e
- Npx 2.0e and 2.0e-beta
- Headstage-64
- Miniscope
- Rhs2116

filcarv, 2024-08-27:
Hi @jonnew 

The tethers are stated in the store, close to each device.

ONIX devices are shipped with the **[Coaxial Tether (X.FL to SMA) micro 8.2 ft (2.5 m)](https://open-ephys.org/onix/oeps-5500)**

Miniscopes are shipped with the **[Coaxial Tether (U.FL to SMA) mini 4.9ft (1.5m)](https://open-ephys.org/miniscope-v4/oeps-5501)**

jonnew, 2024-08-29:
@ChucklesOnGitHub @filcarv The best way to do this is for me to work with you all while you plug in headstages one by one and we test them with multimeter etc. I have installed VS and the code on the ONIX test PC at OEPS. If we can schedule a time to do this let me know.

jonnew, 2025-04-22:
I believe this should be taken care of. I'm closing for now. We can reopen if required.

closed 2025-04-22 by jonnew