open-ephys/bonsai-onix1 #250: Consider adding a step to check for semantic version increase in actions

bparks13, 2024-08-23:
Due to how pre-releases are categorized in semantic versioning, for us to easily test CI builds after a new release is published we need to remember to increase the semantic version in some way (it can be a patch version increase if needed) so that the CI pre-releases can be installed on top of the full release. 

To remove the need to remember this, we can add a step in the Github actions that compares the version number of the current branch to the most recent release and blocks merging a PR unless the version is increased.

This can be done in a number of ways, but one possibility is as follows:
- For current branch, pull out the `VersionPrefix` from `Directory.Build.props`
- For most recent release, grab the tag (`vX.X.X`) and strip the 'v'
- Compare the two versions using a semantic version tester (for example, using `semver` in Python)

This will only check that the major/minor/patch version is increased in some way, since we are utilizing CI pre-release builds in the actions there should not be any conflict with package creation, nor should it interfere with the next release version.

We can also configure it to block the PR but not block subsequent jobs so that the remaining checks run, in case there are other issues that need to be addressed.

jonnew, 2024-08-26:
We discussed during our meeting. I like this. The only thing we came up with is not relying on the tag to create a more recent version to compare against because the tag does not need to be a well formed version number or be incremented correctly etc. We need some way of comparing against an actual previous version rather than the tag which is supposed to be reflection of that, but its not guaranteed.

bparks13, 2024-08-26:
We also decided that the package should never be created if the version number is not increased in some way.

PR #264 (2024-08-27): Add step to check semantic version
- Runs during pull requests to check that the version number is increasing
- Skipped during pushes to main or releases

Fixes #250
closed 2024-08-27 by jonnew