open-ephys/bonsai-onix1 #253: Consider adding a webhook to the repo

bparks13, 2024-08-23:
It is possible to create a channel in discord, and push to that channel whenever specific events happen in Github using [webhooks](https://docs.github.com/en/webhooks). 

I don't think we need to blast every event that happens, but we could make a public channel that receives a push whenever a new release is published so that the server is aware of any updates and can update their libraries if they want to.

Additionally, we can fine-tune the events and push to a private channel (I think? I'm actually not sure if private channels can receive webhooks) just for us to track events.

jonnew, 2024-08-26:
Seems fine. Don't know if its going to have a large impact though.

bparks13, 2024-11-19:
Webhooks have been added for:
- When a release is published, push to the announcements channel in our Discord.
- When a PR is opened / closed, push to the internal update-notices channel in our Discord so the team is aware of current developments.

In the future we can add more hooks if needed, but these are sufficient for now.

closed 2024-11-19 by bparks13