open-ephys/bonsai-onix1 #283: StreamReader arguments in NeuropixelsV1Helper.ParseAdcCalibrationFile() and ParseGainCalibrationFile() should be a path

jonnew, 2024-09-03:
The `file` input argument in the following functions:  

https://github.com/open-ephys/onix-bonsai-onix1/blob/3fdcec67f5f5d548ddf9125d7859565b55a48d08/OpenEphys.Onix1/NeuropixelsV1Helper.cs#L21

https://github.com/open-ephys/onix-bonsai-onix1/blob/3fdcec67f5f5d548ddf9125d7859565b55a48d08/OpenEphys.Onix1/NeuropixelsV1Helper.cs#L65

should either be a `FileStream` or a `string` indicating a path. StreamReader is to general because calibration data is always provided in a file and this function call expects to start from the beginning of the stream. This will require a change in the public API.

bparks13, 2024-09-05:
These methods have been marked as `[Obsolete]` in #291, but will not be modified/removed until 1.0.0.

jonnew, 2024-10-15:
This will be addressed in #304. Im closing since in reality its already addressed in #291.