open-ephys/bonsai-onix1 #294: Add GND as reference option for Neuropixels 2.0

bparks13, 2024-09-10:
Currently the only reference options available for Neuropixels 2.0 probes are `EXT`, and `TIP1-4`. On commercial probes, GND is a valid option for references.

This would only be an option for non-beta probes. We need to go ensure that the correct bits are being turned on/off in the shank config for GND to properly be activated. 

This will only be reflected at the GUI level by an additional entry in the drop-down menu, and all configuration will still be handled by the configuration methods in the base library.

jonnew, 2024-09-16:
@jsiegle In order to do this with 100% certainty, we either need the API source code that IMEC is using or we need a i2c trace of probe configuration when GND is selected as the reference. Can you help to provide either of these?

bparks13, 2025-08-25:
@jonnew 

## Adding GND as a Reference Option

Based on the documentation given by IMEC, it looks like the configuration for using GND as reference can be done by setting the [`referenceBit`](https://github.com/open-ephys/bonsai-onix1/blob/72617ac4eb1832d92fdaa2928d46b2c329142470/OpenEphys.Onix1/NeuropixelsV2.cs#L95) to `3`.

For this library, that can be accomplished by adding an entry to the [`NeuropixelsV2QuadShankReference`
](https://github.com/open-ephys/bonsai-onix1/blob/72617ac4eb1832d92fdaa2928d46b2c329142470/OpenEphys.Onix1/NeuropixelsV2QuadShankProbeConfiguration.cs#L16) to use `GND` as a reference, and then adding a line to the `referenceBit` switch to set it to `3`.

**Question**: What should the displayed name for GND be? Based on the other two names, and how it is represented in the documentation, we have the following options: `GND`, `Gnd`, or `Ground`. Given that we fully write out `External` as an option, I am leaning towards `Ground`.

## Differentiate Between Beta and Non-Beta Probes

The changes above will automatically propagate to the GUI, as it pulls the possible values of `NeuropixelsV2QuadShankReference` for the combo box, so these two changes are sufficient to give users full ability to select GND. However, this presents a complication; the GND option is only available for non-beta probes, and currently there is no distinction between the probe configuration for beta vs. non-beta probes. Specifically, both probes use the same dialog (`NeuropixelsV2eProbeConfigurationDialog`) and the same class (`NeuropixelsV2QuadShankProbeConfiguration`). So, now we need some way to differentiate between beta and non-beta probes, both at the level of the Bonsai editor and the GUI.

Since there is a lot of overlap between the probe configurations, I don't think it makes sense to create a new class to differentiate between beta and non-beta probes, since as of right now the only difference is whether or not GND can be used as a reference.

### Bonsai Editor

Currently in main, there is no way to expand the `ProbeConfiguration` to see/modify the reference, but there is a PR with a [commit](https://github.com/open-ephys/bonsai-onix1/pull/481/commits/a28587893a08c283b9a248d7ecf0b4954b480ba6) that allows expanding of properties such as the `ProbeConfiguration`, so we should be aware of this and assume that in the near future the editor will need a way of filtering.

### Dialogs

The dialogs expose the reference options in a `ComboBox` by calling `Enum.GetValues(typeof(NeuropixelsV2QuadShankReference));`. We can modify this in some way so that if it is a beta probe, then instead it becomes:

```C#
comboBoxReference.DataSource = Enum.GetValues(typeof(NeuropixelsV2QuadShankReference))
                                   .Cast<NeuropixelsV2QuadShankReference>()
                                   .Where(x => x != NeuropixelsV2QuadShankReference.Ground)
                                   .ToList();
```

## Overlap with Other Issues

Since the scope of the issue is increasing to need some way of differentiating between beta and non-beta, there is overlap with #446, which will require re-working the `V2` probe configuration to work with single- and quad-shank probes. I believe it makes sense to work on both of these issues concurrently, by implementing any major changes in the other branch and then using that as a starting point for this issue.

I will expand on my proposal for that issue on the issue itself, but in a nutshell I believe that it will make the most sense to rename  `NeuropixelsV2QuadShankProbeConfiguration` => `NeuropixelsV2ProbeConfiguration`, and add two new readonly properties; `NumberOfShanks` and `IsBeta`. The number of shanks is relevant for #446, but knowing if the probe is a beta probe is relevant for this issue, and could be what is used to decide whether or not to filter the enum list, as shown above.

PR #491 (2025-08-28): Add Ground as reference option for Neuropixels 2.0 probes
Adds `Ground` as an option for the reference. This is added as part of the reference enum, and is propagated to the GUI drop-down menu.

Fixes #294
closed 2025-08-28 by bparks13