open-ephys/bonsai-onix1 #300: Change `BreakoutAnalogInput` to `AnalogInput`

cjsha, 2024-09-16:
This same change can be applied to `BreakoutAnalogOutput`, `BreakoutDigitalInput`, and `BreakoutDigitalOutput`. 

@jonnew mentioned this to me off-handedly that he doesn't like the current naming convention and I think it's worth discussing. It seems redundant, and I'm not sure what the benefit is.

bparks13, 2024-09-17:
This change can actually be done for v0.3.0, since we are utilizing the `[Obsolete]` tag to not break workflows. Especially since we are still in a 0.y.z version and it is considered "unstable". The change that needs to be made for v1.0.0 would be to remove these obsolete classes entirely, since that would break workflows.

bparks13, 2024-09-17:
@cjsha  Correct, they are not searchable, but if they exist in a workflow file they will be loaded correctly. Hence, this does not break existing workflows. There is a visual indicator that they might be obsolete, the color is correct but they do have cross-hatches. I linked an issue in the docs repo stating that we need to update the workflows to ensure that the SVGs on the website don't show this cross-hatching.

PR #301 (2024-09-17): Remove Breakout prefix from classes where its redundant
- Preserve backwards compatibility with old workflows.
- ~This commit requires a major revision increment~ Because we are in the 0 major revision, this change is permitted, even though its API breaking
- Fixes #297 
- Fixes #300 
- After this is merged, I will create 1.0.0 issue requesting the removal of obsolete types
closed 2024-09-17 by jonnew