open-ephys/bonsai-onix1 #306: Add a clock for LFP data in `NeuropixelsV1eDataFrame`

cjsha, 2024-09-19:
In its current implementation, the `Clock` member of `NeuropixelsV1eDataFrame` aligns only with the `SpikeData` member and not the `LfpData` member. 

This ended up happening because the NP1 probe samples LFP data 12x slower than AP data - therefore the `LfpData` buffer is 12x shorter than the `SpikeData` buffer. `Clock` is sampled whenever `SpikeData` is sampled - there is no corresponding clock that aligns with `LfpData`.

One potential solution is to add another member to the `NeuropixelsV1eDataFrame` which is a clock that does align with `LfpData`.

aacuevas, 2024-09-23:
It is true that the LFP in this case is a bit of an outlier in the sense that it is not produced every hardware frame, but every 12. The fact that the `NeuropixelsV1eDataFrame.BufferSize` property is limited to multiples of 12 to account for this and ensure every software frame includes at least one LFP is a proof of this deviation from the usual frames we deal with.

So while Clock is a representation of hardware frames, we should discuss if yet another field is required. This would just be one every 12 values of Clock.

An alternative solution is use Clock and a Bonsai operator that takes one every 12 elements of the array, and include that in the examples for clarity.

jonnew, 2024-10-03:
I think this is wontfix. We just need to provide an example loading script that shows how to do this.