open-ephys/bonsai-onix1 #309: Add 1 second to HarpTime (HarpSyncInputData node)

ChucklesOnGitHub, 2024-09-25:


jonnew, 2024-09-26:
I'll need a bit more info on this one.

aacuevas, 2024-09-27:
It's one of these "what the hardware sends" vs "how it is interpreted".
Harp sends a uint32 value with the current timestamp (in full seconds) each second. But, the latch event for Harp devices to latch this timestamp is after it has already passed. i.e.: Timestamp 0 gets latched when 1 second has elapsed since the start, timestamp 1, 2 seconds from start, etc...

The hardware Harp device on ONIX just reports whatever value it has received from the clock generator. However, any other Harp device reports the values of their internal timestamps, which will then appear to be one second ahead of the one ONIX is receiving.
There are 3 solutions to this, so the timestamps appear aligned:
1- Add a second in hardware. For once, I believe this is incorrect. The hardware should just send what it receives. Since this is an "interpretation" thing, I believe it is the software duty.
2- Add a second to the output on the Harp node. This way the values of the Harp node output and any other Harp devices will appear aligned when just visualizing the outputs.
3- Just add the second in the workflow node used to convert ONI timestamps into Harp-aligned timestamps. At the end of the day, this is what people will be using to use Harp with the system, so it is only at this step where it will be required.

Although I believe 3 is the more "technically appropriate" solution, 2 is probably the one that will confuse the users the less, as they will just see the same timestamps if they just visualize them. So I would just do 2.

PR #319 (2024-10-02): Add one second to Harp time
Fixes #309
closed 2024-10-02 by jonnew