open-ephys/bonsai-onix1 #314: Hide `PolledBno055` properties in GUI dialogs

bparks13, 2024-09-26:
When a `PolledBno055` is under a headstage operator, the `AxisMap` and `AxisSign` variables are not displayed in the Bonsai property pane because these are fixed quantities when paired with a specific headstage. However, when they are added to the default property pane in a configuration GUI, these properties are now shown and editable.

[image]

These variables are not copied back into the main Bonsai window, so there will not be any permanent changes, but if a user changes these values and expects them to make a difference it will seem like a bug.

jonnew, 2024-10-20:
Moving @bparks13 email comments to this discussion

> Due to the accessibility of the type converters, I would have to make every single type converter public, as well as make the base class public, if I want to call the converter directly from the Design library
> 
> It isn’t a huge amount of code, so I could reimplement it, but that is something we are trying very hard not to do
> 
> Question: is it worth exposing all of the type converters in the public interface to make this an easy fix? Or do we just leave it 

> as-is for now, knowing that a user could potentially see these fields and modify them expecting it to do something when in reality it does nothing?
> 
> There is [this page ](https://stackoverflow.com/questions/18926109/c-sharp-net-4-5-propertygrid-how-to-hide-properties)on SO showing it might be possible using reflection, if that is something we wanted to do without making the classes public

My opinions
- I don't think we want to make this public
- Reflection is probably the way to to go
- We can release 0.4.0 without this, so if you think its going to hold things up then lets move to 0.5.0

jonnew, 2024-10-21:
@aacuevas Suggests the following as alternative method without reflection or changing the public interface:

https://stackoverflow.com/questions/920844/how-can-i-access-an-internal-class-from-an-external-assembly
https://learn.microsoft.com/en-us/dotnet/standard/assembly/friend?redirectedfrom=MSDN

bparks13, 2025-08-19:
@aacuevas As I was typing up some notes on a potential way to create a new attribute, and attach that attribute to the `ConfigurePolledBno055` device as a way to propagate the information to the Design library, I realized we have a potentially simpler way to do this.

Since there is a specific Editor/Dialog for each `Headstage` node that contains a `ConfigurePolledBno055`, we can modify the `PolledBno055Dialog` to be constructed with two boolean inputs: 
```C#
public PolledBno055Dialog(ConfigurePolledBno055 configureNode, bool axisMapIsBrowsable = true, bool axisSignIsBrowsable = true)
```

And then we can use the following helper method to modify the Browsability of the individual properties depending on the boolean inputs:

```C#
public static void ChangePropertyBrowsableState(object obj, string propertyName, bool isBrowsable)
{
    PropertyDescriptor descriptor = TypeDescriptor.GetProperties(obj)[propertyName];

    if (descriptor != null)
    {
        BrowsableAttribute browsableAttribute = (BrowsableAttribute)descriptor.Attributes[typeof(BrowsableAttribute)];

        FieldInfo browsableField = browsableAttribute?.GetType().GetField("browsable", BindingFlags.NonPublic | BindingFlags.Instance);

        browsableField?.SetValue(browsableAttribute, isBrowsable);
    }
}
```

In the `PolledBno055Dialog` constructor, it would be called like this:

```C#
DesignHelper.ChangePropertyBrowsableState(configureNode, nameof(configureNode.AxisMap), axisMapIsBrowsable);
DesignHelper.ChangePropertyBrowsableState(configureNode, nameof(configureNode.AxisSign), axisSignIsBrowsable);
```

From the headstage dialog, we would call the `PolledBno055Dialog` with both booleans set to `false`, and this would allow us to hide these properties without needing to add new attributes. 

All of this assumes that each headstage will be consistent with its implementation of the TypeConverter, meaning that we know all instances of `ConfigureHeadstageNeuropixelsV2e` will always hide the `AxisMap`/`AxisSign` properties, and therefore when we open the dialog for this headstage we can set those properties to not be browsable.

I have tested this implementation, and it does work with one caveat; if there is no `Browsable` attribute attached to the  `AxisMap`/`AxisSign` properties in the base class, then the property grid will be completely blank after running the helper method. Adding the `Browsable(true)` attribute fixes this and correctly hides the properties dynamically.

If we like this version, I can finish the implementation and modify how the `PolledBno055Dialog` is called by each headstage to hide the properties as needed.

One other thing to note is that this will not affect how the `PolledBno055Dialog` is displayed if someone places a `ConfigurePolledBno055`  device node by itself, as then it will created by the PolledBno055Editor with the default boolean inputs to mark the properties as browsable.

PR #483 (2025-08-22): Mark AxisMap and AxisSign as not browsable
During our discussions, we came to the conclusion that the `AxisMap` and `AxisSign` variables should not be browsable in any situation; this is different from the original behavior where they were browsable for the device outside of the context of a headstage.

This PR removes the `PolledBno055SingleDeviceFactoryConverter` as it is no longer needed, and marks the two properties as `Browsable(false)`. This removes them from the editor in all instances, which achieves the original goal of hiding these properties in the GUIs.

Fixes #314
closed 2025-08-22 by bparks13