open-ephys/bonsai-onix1 #321: Rename from `ChannelConfiguration` to `ProbeGroup`

bparks13, 2024-10-02:
In many places, the name `ChannelConfiguration` was used for probe groups prior to any channel configurations were created. However, this can be a cause for confusion, and should be renamed for clarity.

An example of an existing place this occurs is below:

https://github.com/open-ephys/onix-bonsai-onix1/blob/de7bb7ef19d2c8c59d78c3a504a701c6ee53bd3a/OpenEphys.Onix1/NeuropixelsV2QuadShankProbeConfiguration.cs#L213

jonnew, 2024-10-03:
`ProbeGroup` is not as clear to me as `ChannelConfiguration`. If the later is somehow incorrect, then lets find a happy medium that describes what the type does. `ProbeGroup` just doesn't give much information IMO.

bparks13, 2024-10-07:
@jonnew The issue in my opinion is that `ProbeGroup` actually describes the object itself; this is how `ProbeInterface` labels the object, and what is currently called `ChannelConfiguration` is actually a probe group. The secondary issue is that we are calling the device level configuration classes `*ProbeConfiguration` to encapsulate all aspects of configuration, including the probe group. 

For me it is confusing to access `ProbeConfiguration.ChannelConfiguration` when I really want to access the probe group. I do see what you mean about `ChannelConfiguration` being more descriptive, but I think it is actually taking away from what the object really is; a probe group contains the channel configuration information _as well as_ the channel locations (even inactive channels), the shape of the probe shanks, as well as metadata associated with the probes. 

Let's keep discussing this before making any changes.

jonnew, 2024-10-10:
I think this is reasonable. Also, maintaining the same naming conventions as the original specifications and API implementation is important for a lot of reasons. So go ahead do this with that in mind.

PR #458 (2025-06-17): Rename channel configuration to probe group for clarity
This PR is a simple rename of most instances of `ChannelConfiguration` to `ProbeGroup`. This name change is to line up with the underlying `ProbeInterface` specification we are using, and to better describe what the variable actually contains. Now, any instance of `ChannelConfiguration` is a dialog that uses an underlying `ProbeGroup` variable to draw the electrodes and their locations, allowing the user to select which channels to configure.

- Fixes #321
closed 2025-06-17 by bparks13