open-ephys/bonsai-onix1 #354: Make it easier to click to select/deselect channels in ChannelConfigurationDialog

bparks13, 2024-10-28:
Due to the current implementation of selecting and deselecting channels, there is overlapping functionality between just click, and click-and-drag. If the mouse moves even the smallest distance while clicking it will generate a click-and-drag event which then has a rectangle that is too small to select anything, and also precludes the user from deselecting since click-and-drag is only used to select channels.

One easy step: 
 - Check for the size of the selected rectangle, and if it is below a certain size then treat the event as a single click, which then gives back the power to deselect via a click event.

Other possible solutions: 
 - Add in a button that needs to be held to deselect (i.e., <kbd>Shift</kbd> + click), which can be for <kbd>Shift</kbd> + click-and-drag as well to bulk deselect channels

bparks13, 2024-10-30:
Upon further testing, it appears that if there is more than one probe listed (such as in `Rhs2116ChannelConfigurationDialog`) the offset of the first probe is not taken into account when grabbing the index of the contact. This is an error, and should be corrected by applying the offset of all probes to the current index.

jonnew, 2024-11-14:
I did some integration testing and everything feels very nice. Not extensive though.

PR #373 (2024-11-14): Make channel selection smoother for all channel configuration dialogs
If multiple probes exist, ensures that the contact index is correctly grabbed.

Better differentiates between clicks and click-and-drag events so it is easier to select / deselect contacts.

Fixes #354
closed 2024-11-14 by jonnew