open-ephys/bonsai-onix1 #360: Change "Configuration" label for aggregate operators in properties panel?

cjsha, 2024-10-29:
[image]
To me, this "Configuration" label suggests that the `Name`, `Port` & `PortVoltage` properties are configuration properties. I'm not sure if this was intentional. "Aggregate" or "Hub" (though I think we are hiding that term from users) might be more correct labels? In any case, it happens to be true that all those properties are configuration properties, and I'm not sure there will ever be any acquisition properties that would fall into that section of the properties panel anyway. This is a nitpick.

jonnew, 2024-10-30:
I think it's correct. These are configuration properties in the sense that they have no effect at runtime. In the same way that device level configuration properties don't.

cjsha, 2024-10-30:
Sounds good

closed 2024-10-30 by cjsha