open-ephys/bonsai-onix1 #383: ReadBufferSize should be automatically calculated

aacuevas, 2024-12-11:
Since we know the bandwidth each device requires, it should be possible to calculate an optimum ReadBufferSize. The StartAcquisition operator could just check which ConfigureX operators are there, which devices are enabled and set an optimum value for the buffer.
The parameter would still exist, just blank (similar to the voltage parameters). If filled, then this value will override the automatic setting, useful for users that might want to fine-tune the experiment.

cjsha, 2024-12-12:
I'm curious: how would we take into account event-based data streams (i.e. digital inputs) which don't have a fixed data rate?

jonnew, 2024-12-17:
@cjsha thats a good point. The automatically calculated size should be pretty generous. In the case where there is an extreme amount of data coming in, they would need to revert back to setting the size manually. The same is true if they want to absolutely optimize closed-loop response, but in the other direction.

I gave this a shot and I ran into an issue I need to try to understand. Everything works well if I only configure the acqusition board. However, if I add a neuropixels headstage, this line is called only once and the Max read and write sizes are as if the neuropixels was not there. I have a feeling this is because the device table itself is not modified with a pass through device, and therefore something occurs out of order here.

[image]

The result, predictably, is that the read size is too small for the neuropixels data:

[image]

This is more a note for myself to track down what's going on tomorrow.

jonnew, 2024-12-17:
After talking through this with Aaron, we came to the conclusion that doing a constant multiple of Max Read and Write Size is such a hack, and so prone to not functioning in a variety of circumstances, that we will wait until the proper information is made available by the prosed modifications to device datasheets in https://github.com/open-ephys/ONI/issues/65.

After this we can revisit this issue by adding functionary to all classes that derive from `SingleDeviceFactory` that takes information about their enable state and specified bandwidth into account to determine the overall bandwidth of the device table. @aacuevas  has created lookup table for the required Block sizes for difference bandwidths that can be used for the basis of a lookup table.

jonnew, 2025-04-21:
To do this properly requires implementation of the ONI2.0 specification. However, the change that this would entail is a modification of each device's datasheet to include expected bandwidth for read and write streams. This can be done currently even though this library is working against ONI1.0 hardware currently.