open-ephys/bonsai-onix1 #390: Why does `Rhs2116StimulusTrigger` take "double" instead of "bool"?

cjsha, 2024-12-18:
I'm wondering why `Rhs2116StimulusTrigger` takes "double" instead of "bool" as the type of its input observable.

For reference, `Headstage64ElectricalStimulatorTrigger` and `Headstage64OpticalStimulatorTrigger` take "bool" as their input observable type.

jonnew, 2025-01-06:
The trigger provided to an `Rhs2116StimulusTrigger` defines a delay in microseconds that will be applied prior to the defined stumulus. 
https://github.com/open-ephys/bonsai-onix1/blob/3c6702bd498774dcef140aa23998b06123730da8/OpenEphys.Onix1/Rhs2116StimulusTrigger.cs#L37

This needs to be documented properly:

1. Explain the difference between per channel Delay property and this global sequence delay.
2. Update the RHS2116Trigger `TRIGGER` description with this information

[image]

cjsha, 2025-01-06:
My follow-up question is: should hs64 trigger operators follow the same pattern? They have a delay property instead of taking in delay in their input sequences

jonnew, 2025-01-13:
If we want to do this, we need a firmware change. Since this device is already in the field, we wont fix.

closed 2025-01-13 by jonnew