open-ephys/bonsai-onix1 #395: Rename Rhs2116 stimulation parameters

cjsha, 2025-01-02:
I propose:
- `Inter-Pulse` --> `Intra-Pulse`
- `Inter-Stimulus` --> `Inter-Pulse`
- `Define Stimuli` --> `Define Stimulus`

[image]

glopesdev, 2025-01-02:
@cjsha I feel these terms might have defined meanings in the community which go quite far back in time, and so it might be confusing for people if we change them.

A good reference to consider here might be the [PulsePal Wiki](https://sites.google.com/site/pulsepalwiki/parameter-guide?authuser=0), since PulsePal is also listed on the Open Ephys website. Below I paraphrase their key terms:

- A "pulse" is a single deflection from resting voltage, and can be "monophasic" or "biphasic";
- The "inter-pulse interval" is the time between consecutive pulses;
- For biphasic pulses you can control the time between phases by setting the "inter-phase interval";
- A "pulse train" is the entire sequence of pulses delivered by the stimulator on a trigger;
- The "pulse train duration" is how long the entire sequence lasts.

To create complex pulses, PulsePal also allows defining "bursts", which are short sequences of pulses repeated within the pulse train. A burst has its own parameters:

- The "burst duration" defines how long each of these short sequences lasts;
- The "inter-burst interval" is the time between consecutive bursts, and allows spacing out the short sequences over the entire pulse train.

I'm not entirely sure whether what is currently called here a "stimulus" corresponds to a burst or a pulse train, but I imagine probably the former, since you can specify an "inter-stimulus interval".

Happy to discuss further, as I think it would be really nice if nomenclature could be aligned across the different projects in the ecosystem.

cjsha, 2025-01-02:
> I'm not entirely sure whether what is currently called here a "stimulus" corresponds to a burst or a pulse train but I imagine probably the former, since you can specify an "inter-stimulus interval".

> P.S.: On a second reading I think maybe Define Stimuli here refers to "Pulse train" and Inter-Stimulus refers to the "inter-burst interval", in PulsePal terms.

According to this [Rhs2116 GUI guide](https://open-ephys.github.io/bonsai-onix1-docs/articles/hardware/rhs2116/gui.html), "inter-stimulus" currently refers to the time between pulses, so I think my proposed change ("inter-pulse") is in agreement with what you would want. "inter-pulse" currently refers to the time between anodic & cathodic phases of the pulse which is so I think "inter-phase" could work well instead of my proposed "intra-pulse". In any case, both "inter-stimulus" & "inter-pulse" labels seem misleading as they are.

Also the RHS2116 capability is not 1:1 to pulsepal- afaik there is no bursting capability.

bparks13, 2025-01-02:
RHS2116 has more capabilities than we can currently utilize with the GUI. As of right now, it is limited to a single "burst" per channel. This means that there is a constant time between all stimuli that cannot be currently modified. However, the hardware does allow for an arbitrary timing between successive stimuli, which we plan to add functionality for in the future.

Regarding the initial proposal, the only one that I think might be valid is changing the term to `Define Stimulus`, since the definition of a `Rhs2116Stimulus` in the code is the totality of the waveform delivered as stimulation for a single channel. However, my hesitation with doing that is that we have a field for `NumberOfStimuli` inside of `Rhs2116Stimulus`, where it is describing the number of paired positive/negative pulses for the channel. Therefore, changing the term in the GUI would only be correct if a single stimulus is delivered per channel. I'm happy to keep discussing this terminology though, since we should be consistent with how we discuss these terms.

The other two terms are applied correctly, and since @cjsha linked the GUI docs I just wrote I have found an error in how I described the setting. Internally, a "pulse" is defined as _either_ an anodic or cathodic (positive/negative) stimulation, whereas a "stimulus" is a paired set of two pulses. Therefore, `Inter-Pulse` is the time between successive anodic-to-cathodic (or vice-versa) pulses, and `Inter-Stimulus` is the time between successive positive/negative pulses. Changing them as proposed would be incorrect. These terms do need to be corrected in the docs though, but we can wait to push that change until we update the images for the `ProbeInterface` defining the explicit orientation as proposed in #391.

cjsha, 2025-01-02:
I see. Then this is an issue that should exist in the docs repo then. I made an [issue in the docs repo](https://github.com/open-ephys/bonsai-onix1-docs/issues/175) and I will close this one.

closed 2025-01-02 by cjsha