open-ephys/bonsai-onix1 #397: Thread deadlock encountered when using BNO055

bruno-f-cruz, 2025-01-10:
Heya!

Following a chat with @bparks13...

When using the library with a custom Bonsai workflow ([repository is here](https://github.com/AllenNeuralDynamics/Aind.Behavior.Pirouette/blob/f7e1dae88f5af9988e646d9d075a25f590272605/src/Pirouette.bonsai), but let me know i you have any questions), we have encountered a thread deadlock after running long acquisition sessions (>4hours). From the point of view of bonsai, the lock manifest itself as something that stops the update of all visualizers and prevents the "stop" button from being pressed.

After making a quick memory dump from the process while in its frozen state, I realized that one of the threads that was locked was the one used by the PolledBNO055Data operator. (I am happy to provide the mem dump if you want, but its quite a large file (2gb).)

In the meantime, I tried a solution that seems to have fixed it (or at least no crashes so far, including a full 12hour, 0 twisted turns on the wire, session :P)

Going through the source of the operator (https://github.com/open-ephys/bonsai-onix1/blob/3c6702bd498774dcef140aa23998b06123730da8/OpenEphys.Onix1/PolledBno055Data.cs#L57), I noticed that the 0-argument overload is instantiating a NewScheduler. As I have had trouble with these before, I instead used the second overload with a Timer (which by default uses the DefaultScheduler, if I am not mistaken). This change has been sufficient to stop the deadlocks so far.

Hope it helps!

jonnew, 2025-02-25:
Hi Bruno. Just looking into this now. The reason this, at least according to my note above that line is:

> // NB: NewThreadScheduler runs polling on dedicated thread and uses StopWatch for high resolution
> // periodic measurements. This results in much better performance than the default scheduler. 

So I guess this issue might be pretty deep within .NET or the OS. BNO data will always be hardware timestamped when its received by the host, so you wont loose resolution on marking when a reading was taken. However, going with the `Timer` option will limit the readout speed pretty severely.

bruno-f-cruz, 2025-03-24:
Heya @jonnew. 

After further testing, I am not sure if there is a problem with the BNO acquisition anymore. We found a configuration that reliability leads to freezes even with the solution I mentioned above. We have since removed a recursive pattern in the workflow that has seemingly eliminated the crashes. While the locked threads were indeed between the BNO and the commutator, it appears the cause may be related to .net/reactive framework and the likely anti-pattern we were trying to use. I will close this issue for now to not lead people astray!  

Thanks for taking a look!

closed 2025-03-24 by bruno-f-cruz