open-ephys/bonsai-onix1 #404: Breakout Board indicator lights not intuitive

ChucklesOnGitHub, 2025-01-20:
Digital IO and Acquiring indicator LEDs seem to be opposite as intended.
Digital IO are on when low and and Acquiring  is on when idle, which to me is counter-intuitive and also does not match the docs:

[image]

jonnew, 2025-02-22:
- DIO: They are not on when low, the lines are pulled up when open. If you ground one of the digital inputs, the corresponding LED will turn off. 
- Acquiring: Hmm, I'll need to look into it.

jonnew, 2025-04-21:
These illumination states are controlled by the breakout gateware. Putting in a fix is probably not practical. We do need to update the docs to 

1. Fix the the Idle/Acquiring led states
2. Explain that the DIO is pulled high by default.

jonnew, 2025-04-28:
See this PR: https://github.com/open-ephys/onix-docs/pull/97

closed 2025-04-28 by jonnew