open-ephys/bonsai-onix1 #412: Fix or Remove Broken links to ONI documentation in XML Comments

cjsha, 2025-02-21:
This is the output from lychee link checker when building the bonsai-onix1 docs:

 ## Errors per input

### Errors in _site/api/OpenEphys.Onix1.ContextTask.html

* [404] [https://open-ephys.github.io/ONI/hw-spec/dev_table.html#dev-address](https://open-ephys.github.io/ONI/hw-spec/dev_table.html#dev-address) | Network error: Not Found
* [404] [https://open-ephys.github.io/ONI/hw-spec/devices.html#dev-desc](https://open-ephys.github.io/ONI/hw-spec/devices.html#dev-desc) | Network error: Not Found
* [404] [https://open-ephys.github.io/ONI/hw-spec/controller.html#data-frames](https://open-ephys.github.io/ONI/hw-spec/controller.html#data-frames) | Network error: Not Found
* [404] [https://open-ephys.github.io/ONI/hw-spec/dev_table.html](https://open-ephys.github.io/ONI/hw-spec/dev_table.html) | Network error: Not Found

### Errors in _site/api/OpenEphys.Onix1.CreateContext.html

* [404] [https://open-ephys.github.io/ONI/api/liboni/driver-translators/index.html#drivers](https://open-ephys.github.io/ONI/api/liboni/driver-translators/index.html#drivers) | Network error: Not Found

### Errors in _site/api/OpenEphys.Onix1.StartAcquisition.html

* [404] [https://open-ephys.github.io/ONI/hw-spec/controller.html#data-frames](https://open-ephys.github.io/ONI/hw-spec/controller.html#data-frames) | Error (cached)

bparks13, 2025-02-21:
Ah, this makes sense, we recently pushed some changes to the ONI documentation that adds a version selector. The downside of this change is that it modifies the hierarchy of the website so instead of `ONI/hw-spec/devices.html` it becomes `ONI/main/hw-spec/devices.html`. The relevant question here is which version of the ONI docs we want to target here.

My guess is that we want to point it towards the stable version, meaning we need to modify the URLs to point to `ONI/v1.0/*` instead of just `ONI/*`. @jonnew @cjsha @aacuevas any other comments?

aacuevas, 2025-02-21:
Yes, the current library uses the existing API and specification, so it should link to the v1 docs.

cjsha, 2025-02-21:
Per our discussion today: remove links to specific sections of the ONI docs, Only keep the one that has a functional redirect (the first link [here](https://open-ephys.github.io/bonsai-onix1-docs/api/OpenEphys.Onix1.ContextTask.html)) and put that on each page that currently has links to ONI docs.

jonnew, 2025-02-22:
@cjsha was this fixed in #415? If so, please close this and also remember to put `Fixes` in your PR description next time.

closed 2025-02-24 by cjsha