open-ephys/bonsai-onix1 #416: Default Channel configuration for Headstage64 RHD2164 device

dweiss044, 2025-02-24:
Hello,

I'm wondering is there a default channel mapping for the RHD2164 device on a Headstage64 chip? I had a channel map that was working on the previous version of Bonsai.Onix[https://open-ephys.github.io/onix-docs/Software%20Guide/Bonsai.ONIX/index.html](url), but now that I upgraded to the new Bonsai.Onix1, my channel map is scrambled.  I tested the channel configuration by injecting a small amount of noise through saline into half of the channels, and it seems like the Intan chip's channels don't correspond to the Bonsai channels. 

A little more info: I'm using an [eib-neuronexus-a64](https://github.com/open-ephys/eib-neuronexus-a64/blob/main/eib-neuronexus-a64.kicad_sch) adapter, which has 2 32-channel buses for neuronexus probes.  The top bus is connected to all even RHD2164 channels, while the bottom bus is connected to all odd channels, via the Headstage-64. However, when I connect to only 1 bus and put through a test signal, every other odd channel and every other even channel reflects the signal, which is inconsistent with the hardware wiring. Is there a default remapping in software that I should be accounting for? 

Thanks,
David

bparks13, 2025-03-03:
Hi @dweiss044, thanks for pointing this out! Upon review, we discovered that there was a remap done in the software with the old library, which was not replicated in the new one here. You can see the remap and the details from the hardware specification here: #422. 

Let me know if this makes sense, and we can revisit this channel mapping once the fix has been implemented to see if that is sufficient for you. If not, we have been working with the ProbeInterface [specification](https://probeinterface.readthedocs.io/en/main/overview.html) to be able to handle device channel indices across different adapters, and I can help with that if we need to take that step later.

dweiss044, 2025-03-03:
Hi @bparks13, thanks for the help! I reordered according to the map in #422 and that solved the issue.  I thought ProbeInterface specifications can only be used for Neuropixels and RHS2116 headstages for now, is that correct? Is there a way to use them for Headstage 64/RHD2164 devices?

bparks13, 2025-03-03:
@dweiss044 Great! I'm glad to hear that ordering seems to work, we will definitely make sure to get the software remapping done so that in the future no extra steps are needed on the user end.

For now, you are correct. We have not implemented any GUIs for Headstage64, which is where we added functionality for the ProbeInterface files in the other devices. We plan to tackle this soon, but I can't guarantee when the GUI will be added.