open-ephys/bonsai-onix1 #423: Export metadata next to data files

bparks13, 2025-03-05:
When a recording is started, from a processing pipeline perspective it would be helpful to have a JSON file that is created next to the raw data file that contains metadata such as:

- Number of channels 
- Sample rate

For Neuropixels, this metadata can be extended to include other things, but many of these NPX specific metadata can currently be manually exported to a JSON file by following [these instructions](https://open-ephys.github.io/bonsai-onix1-docs/articles/hardware/np1e/gui.html#save-probeinterface-file). This would include things like: 

- AP / LFP gain
- Currently enabled electrodes
- Filters
- etc.

This could be incorporated into #385 as well. Additionally, we should also consider moving the probe interface configuration information outside of the *.bonsai file as proposed in #366.

bparks13, 2025-03-05:
@jsiegle @anjaldoshi @medengineer Here is the issue we discussed, please let me know if there are any additional considerations we should be aware of,

jonnew, 2025-03-19:
Per our discussion today, and others, we feel that the solution to this issue should come in two steps

1. Address #247. The reason for this is if the way that ONIX data is save is constrained, then the problem of documenting metadata is narrowed. This is especially true if data produced by this node is somehow marked with the fully-qualified device address that produced the data.
2. To address this issue, create some type of .bonsai file parsing library to extract metadata. It is contained within the file and therefore producing secondary files feels highly redundant.