open-ephys/bonsai-onix1 #430: headstage lock check sometimes fails spureously

aacuevas, 2025-04-08:
Sometimes, in a hard to consistently reproduce way, headstage link detection fails with the "Unable to obtain lock" message.
When this happens:
- It is immediate (i.e.: no actual port sweep seems to be taking place)
- It might happen even if a fixed voltage is set

So it might be some _other_ actual exception that just occurs on the startup process and this block, in turn, throws the unable to obtain lock one, or some locking issue.

A couple more details:
- When this happens, pressing Bonsai's restart workflow button seems to trigger this issue more reliably, which suggest some cleaning up issue or timing-related issue
- We have observed this on HS64, not sure yet if this happens with other headstages as well
- Workflow topology (how many operators, visualizers, etc...) seems to affect the probability of this happening, which also points to a deferred resource disposal clash or similar issue.

jonnew, 2025-04-28:
@ChucklesOnGitHub I'm having trouble reproducing this. Can you share a workflow and hardware setup that most reliably causes the problem?

ChucklesOnGitHub, 2025-04-30:
Here is a video. In this instance the issue happened only a few times, and only on pressing Restart, but not Stop and Play.

I've seen instances of this happening 1 out of 2 times (when the user starts the workflow, it fails, they ignore the pop-up and click Start again) or even failure every time for a long streak until it finally worked. There are no hardware changes between restarts.

[issue430.zip](https://github.com/user-attachments/files/19981561/issue430.zip)

jonnew, 2025-05-23:
I think I've found the cause of this issue. 

During a restart, the headstage is turned off and then on again very rapidly. This seems to cause the serializer not to be completely shut down and to be in an indeterminate state. To combat this, I've simply added a delay before the first voltage to turn the headstage on is applied.

PR #450 (2025-06-02): Add 1 second delay before applying port voltage to headstage-64
- During a restart, the headstage is turned off and then on again very rapidly. This seems to cause the serializer not to be completely shut down and to be in an indeterminate state. To combat this, I've simply added a delay before the first voltage to turn the headstage on is applied.
- Fixes #430
closed 2025-06-02 by aacuevas