open-ephys/bonsai-onix1 #431: Polled Bno: Acceleration data shows jumps when centered around zero

bparks13, 2025-04-10:
This behavior was noticed when testing the Neuropixels 2.0e headstage, but presumably is present for any Polled Bno devices.

When streaming acceleration data, if one of the axes is around zero there are strange jumps in the data periodically that always go to about the same value, plus or minus (see image below). There are no exceptions thrown by the library, so this does not appear to be an abnormal register read, but might be due to how the register value is handled.

[image]

bparks13, 2025-04-10:
See if this issue can be reproduced in the regular `Bno055` device

jonnew, 2025-04-21:
The issue is the following: the `PolledBnoData` class reads BNO055 data 1 byte at at time. If an update occurs in between reading the MS and LS bytes for a particular value (in this case acceleration), you can malformed 2s-complement numbers resulting in these jumps around zero. Although they are most obvious around zero, its actually a general problem.

The correct solution is to read words larger than 1 byte per I2C cycle. However, the current gateware implementation does not allow this. The register interface provides a 32-byte interface so there is not a fundamental limitation. The gateware needs to be updated to allow 32-byte I2C access. This is probably the correct solution because it does not involve hacks and would improve the overall performance of Polled I2C access in general, potentially obviating the need to only access particular registers to increase polling frequency as is done here:

https://github.com/open-ephys/bonsai-onix1/blob/cdc835e882bc61d5f3121f3d56020401e5c298bd/OpenEphys.Onix1/PolledBno055Data.cs#L172

closed 2025-06-24 by jonnew